moode-player/moode #8: Cover backdrop not working

duenni, 2018-10-26:
Hi,

I try to enable cover backdrop but nothing will happen. moode refreshes after the option is enabled under "Customize" but cover backdrop is not there. Going into the "Customize" menu again shows the options are blank again. Clearing browser cache does not help. I also tried something like clear cache, set options, clear cache again to no avail. Tried rebooting after setting the option, still nothing. Tried with Firefox 63, Chrome 70.0.3538.77, Safari 12.0.

moodeaudio, 2018-10-26:
Post /var/log/moode.log and I'll have a look.

moodeaudio, 2018-10-26:
Odd ip address 192.168.178.207.

Is this a home or work network?

duenni, 2018-10-26:
This is a normal private IP from `192.168.0.0/16` in a home network.

duenni, 2018-10-26:
Well, after disabling/enabling "local UI display" and a reboot, no cover art for radio stations is displayed anymore at all.