moode-player/moode #9: Enhancement: Multiple sort options for album view

cortices, 2018-11-30:
Feature suggestion to add an option to allow changing the sorting method of the album cover view without going into Customize settings.

This is useful for those (like myself) with an eclectic music library, who often change whether sorting by Album Artist or Album, (and by extension Year, Genre, Composer, etc.)

This might suggest also repopulating the sub-title label for each album with the relevant metadata to make it clear where sorting changes happen (e.g. displaying the release year instead of Album Artist when sorting by year).

moodeaudio, 2018-11-30:
Hi, the moOde project has centralized all support, feature requests and discussion in the moOde Forum http://moodeaudio.org/forum

If you wouldn't mind reposting your nice feature requests there it would be greatly appreciated.

-Tim

moodeaudio, 2018-11-30:
The moOde project uses Github for source code archiving and release management, and it works beautifully for these two activities. We don't use it for support because the majority of our users are not software developers and don't have Github accounts. They are mainly computer audio enthusiasts that are familiar with, and used to Forum based support for example diyAudio. 

Therefore since our project has to provide a discussion Forum for our users it makes sense to also do all support out of the Forum. The alternative would be to ask our users to also create Github accounts for any issues that may be posted in the repos.

As far as the repo issue section goes I see no reason to close it because its still useful in the event there are issues posted regarding the org or repos themselves or suggestions on how to improve them.

-Tim

cortices, 2018-12-01:
I am happy to do so.

closed 2018-12-01 by cortices

moodeaudio, 2018-12-01:
Great, thanks. 

Feel free to consolidate your ideas into a single Forum post :-)

One of your observations regarding "same album name" being incorrectly rolled up has already been addressed in upcoming moOde 4.4 release :-)

-Tim