moode-player/moode #15: Audio scrobbler won't scrobble when using UPNP Renderer/Tidal

Yonn-Trimoreau, 2019-01-13:
Is it an attended behavior?

The userid and password in the audio scrobbler config screen must be the same as my last.fm login username and password? Or do I have to create API keys?

Do I have to install some system libraries for this functionality to work?

Where are the concerned logs located and what should I look for in order to debug this functionality?

moodeaudio, 2019-01-13:
Hi,

Your best bet is to post this question in the moOde Forum in the support section. There are several users that are using Last.FM with moOde and maybe they can help you out :-)

http://moodeaudio.org/forum

-Tim

Yonn-Trimoreau, 2019-01-14:
Thank you Tim.

http://moodeaudio.org/forum/showthread.php?tid=905

closed 2019-01-14 by Yonn-Trimoreau