moode-player/moode #22: No audio output on a pi 3B

Wqrld, 2019-03-08:
I just installed a new machine on the newest version of moode but im not getting any audio output using the inbuilt soundcard, what would i need to turn on?

moodeaudio, 2019-03-08:
post the output from moode log and i'll have a look

cat /var/log/moode.log

Wqrld, 2019-03-08:
https://paste.imluc.ga/puvaxewegi.sql @moodeaudio

Wqrld, 2019-03-08:
seems like the alsa speaker-test is also not working

Wqrld, 2019-03-08:
fixed by downsampling the alsa audio to mono and rebooting

moodeaudio, 2019-03-08:
The log output shows 

ALSA mixer name (Mic)

The mixer name for the on-board audio device is 'PCM'.

This suggests something external to moOde software.

Wqrld, 2019-03-08:
Coming from mopidy, i really liked the tunein and spotify support on the web ui, does moode also have a  integration for one of those? (i know you offer spotify connect but that would be a cool extra)

moodeaudio, 2019-03-09:
Our approach is to implement the audio renderers and make use of their streaming protocols to play from the native Music Service applications. It's been my experience that the native apps are always much better than integration attempts.

Are u using the new moOde 5 beta?

Wqrld, 2019-03-09:
> Our approach is to implement the audio renderers and make use of their streaming protocols to play from the native Music Service applications. It's been my experience that the native apps are always much better than integration attempts.
> 
> Are u using the new moOde 5 beta?

no i'm using the public moode 4, is 5 already public / feature list? 
I'm likely going to try to run mopidy alongisde of moode for the best experience

Wqrld, 2019-03-09:
I'm also only able to skip songs and not use new ones from other playlists, if this requires spotify premium where is the config file?

Wqrld, 2019-03-09:
i just installed mopidy next to this for the best experience, keeping this open as a suggestion

moodeaudio, 2019-03-09:
moOde 5 beta
http://moodeaudio.org/forum/showthread.php?tid=1183

Spotify track/skip issue is fixed in this beta.

closed 2019-03-29 by moodeaudio