moode-player/moode #23: SyntaxError: At least one digit must occur after a decimal point

piotr277, 2019-03-26:
With Moode 4.4 the player UI is not displayed correctly with my Blackberry Playbook. No reactions on clicking, only blank page is displayed:
[image]
After some digging in I've found that there is problem in playerlib.js which may cause these problems:
[image]

pccasto, 2019-03-26:
Did you use your Blackberry Playbook to connect to a previous version of moOde? What you are seeing (visually) looks a lot like the symptoms of having cached css and js files from a previous version.

Can you clear the cache and try again? I would expect the syntax/reference error to clear.  The bgimage not found might continue, but that is harmless.

The problem of stale cache files on upgrade has been fixed in 5.0beta.

~paul

piotr277, 2019-03-27:
> Did you use your Blackberry Playbook to connect to a previous version of moOde? 

No, I didn't. I did the upgrade from 4.3 to 4.4 before I used Playbook.
I've also cleared the cache but it doesn't help.

The `SyntaxError: At least one digit must occur after a decimal point` happens in `playerlib.js` in this line: `const FEAT_ADVKERNELS =  0b0000000000000001;	//     1`

monkeypox, 2019-03-27:
It's weird that line would be a problem since earlier moodes also used the same construct, your console might be ignoring comment lines, if you don't have a way to inspect it try adding something like...

var test123 = '';

...right after that line, if it really is the problem then the error line number will stay the same otherwise it'll shift by 1. If it does shift you can use the same method to find the actual line by moving you put the var test line down the source and stopping when the line in the console doesn't shift.

pccasto, 2019-03-28:
What version of JavaScript does your playbook browser use?
A literal "0b[01]+" is valid in ECMAScript 6, and is interpreted as a binary value
So ```const FEAT_ADVKERNELS = 0b0000000000000001;``` would be valid with that version.

I'd expect your version might predate ECMAScript 6 (pre 2015), so may not handle that syntax.
That would explain why it is throwing an error message when it tries to parse that line.
It expects a value starting with 0 to be numeric, and throws an error message that isn't completely accurate (there isn't a decimal point in the string), but that's the only error it has when it has a mix of numeric and alpha values.

You may be able to modify the binary values to decimal in the assignments, and have things work in your local copy.

From a coding standpoint, and based on the way these values are used, it is much more readable to express them as binary fields. It may not be worthwhile to convert them in the main line, considering the majority of systems will have ECMAScript 6.

~paul

piotr277, 2019-03-28:
Thanks for your feedback.
I've just done a 'clean' installation of Moode 4.4. Just to eliminate 'the upgrade' aspect.
And it's still the same.
Here are some details I've grabbed about Playbook's browser: http://html5test.com/s/07244543debc1edb.html

I'm pretty sure that lines 116-127 of playerlib.js cause the problem. I will try to edit them but I'm not sure where the `playerlib.js` is located. Is it rendered by [playerlib.php](https://github.com/moode-player/moode/blob/r44prod/www/inc/playerlib.php)?

I know that Blackberry Playbook is quite legacy kind of device, but it's still capable of doing some multimedia & smart home related tasks. And it looks as quite nice candidate for UI panel of Moode.

chris-rudmin, 2019-03-28:
@pccasto A possible fix for browsers which don't support binary literals would be to use `parseInt` or just replace the value with the decimal representation. eg. `0b0000000000000010` is equivalent to decimal value `2` which is also equivalent to `parseInt('10',2)`

monkeypox, 2019-03-28:
Playerlib.js is in /var/www/js/

We just addressed another issue related to using arrow functions which are ES6 and not implemented into WebKit until later. I think the WebKit team was fairly early on support for flexbox and css variables so fingers crossed.

moodeaudio, 2019-03-29:
The feature availability bit mask was designed to control what major features (Airplay, Squeezelite, etc) appeared in the UI by comparing a particular FEAT_ constant with a global SQL value.

The whole thing could be deleted cos no usage scenario has yet come up where it would need to be used.

-Tim

pccasto, 2019-03-29:
Be prepared for one step forward, new obstacle found :-). Not to say it isn't worthwhile, but it may take a bit of digging to sort everything out.

I did a comparison of errors returned from ```jshint``` configured for '"esversion": 5' and '"esverision": 6' to see what other pitfalls might exist.  There is a lot of noise, but thanks to ```diff``` I was able to get a small delta:

Other than the inability to handle binary assignments, this was the other difference between ES5 and ES6:  ```playerlib.js: line 2423, col 21, 'Uint16Array' is not defined.```
So there will need to be some adjustments around that line to handle the Array construct. I didn't look to see how complex that change would be, yet.

I deleted the variables defined as binary, and found that ```FEAT_UPNPSYNC``` is used one place, so simply removing those lines would create an error when that line is encountered. Either that variable will need to be defined using one of the methods suggested above (decimal, or parseInt), or the line where it is used would need to be modified.

There are a lot of places where jslint throws warnings for both ES6 an ES5.  The code works, in spite of those warnings for most browsers, however, some of those warnings might cause an older scripting engine some troubles.  Also, I just ran this against one of the files (playerlib.js).

I think the best course of action might be to do some clean-up of files based on jshint to first ensure they are clean from an ES6 perspective. Make sure that doesn't break anything. Then evaluate any changes that might be needed for ES5 compatibility. Timing/prioritization of that effort would be up to others. But I'd be OK taking it on post 5.0, if that makes sense.

piotr277, 2019-03-30:
Thank you guys for your advices. I've changed const values to decimal and voila! - it works 😃

pccasto, 2019-03-30:
Great.  The other line I identified as a problem is within a handler that is called when clicking to get a random album.  My advice..., don't use that particular feature, and you will be fine for now :-).

monkeypox, 2019-03-30:
It’d be worth trying it to see if it does break, the iPad 3 I have that choked on arrow functions doesn’t have a problem with the random album function and the uint16array. Iirc that part of the code was part of some boilerplate for using the crypto random number generator.

Compatible within reason is a good goal, but if it’s something like lack of support for css variables or flex box then that’s probably a case for a separate branch.

pccasto, 2019-03-30:
True. More info is worthwhile. I was being a bit flippant.  ("Doctor, it hurts when I do this." ... "Well, don't do that any more.")

Interestingly enough, the html5test.com result that was attached earlier shows that Typed Arrays **are** supported by this browser, even though that is an ES6 feature.  So maybe it will work.

@piotr277 - thanks for getting that report as part of this investigation/troubleshooting.  That is the type of info that is very helpful in understanding the different perspectives on the problem.

pccasto, 2019-05-03:
Labeled as bug & enhancement -- depends on perspective. Bug from user point of view, enhancement from requirement POV -- do we support older browsers, or only ES6 and above.  In this case, the change to allow for this support is easy, and still ES6 compliant - it just doesn't take full advantage of ES6.

I'd expect that this particular problem would occur durring a code linting or refactoring of javascript somewhere in the 5.x time-frame.  But meanwhile the impacted user has a means of editing the source in his own environment.

moodeaudio, 2019-06-13:
We can reopen if more trouble reports.

closed 2019-06-13 by moodeaudio