moode-player/moode #25: Screenshots

ShalokShalom, 2019-04-02:
Including screenshots on your homepage might be a good idea to avoid forcing users to install the application just in order to take a brief look onto the UI. 

Or let them google for (old and outdated) screenshots somewhere on the Internet.

moodeaudio, 2019-04-02:
No disagreement on my part that high quality documentation and product pictures are a valuable asset to any project.

Maybe you would be willing to help our project and take on this task ??

-Tim

Norde, 2019-04-14:
I'll do some screenshot (with description) of incoming 5.0 release :)

I think using a slideshow in place of actual moode 4.4 background could be fine. (http://moodeaudio.org/img/moode-r44.png)

moodeaudio, 2019-04-17:
I added a Lightbox w/screen shots to the Features section.

Norde, 2019-04-19:
Sorry for not being helpful 😞

ShalokShalom, 2019-04-20:
All fine. I like it that way. :)

closed 2019-04-20 by ShalokShalom