moode-player/moode #28: Branch Creation Permissions

pccasto, 2019-05-02:
Cloned the repository, created the develop branch in moode-player/moode, and attempted to push it back from the CLI, but was unsuccessful. However I was able to go into the GitHub interface, and create the develop branch.  So that branch does exist now.

I then tested the commands in my fork, and was able to use the CLI as expected. So this may be a team-permissions issue.

I'm not sure if this will cause issues for our workflow, but am still experimenting.  Details are below:

Unsuccessful attempt to create the develop branch in moode-player/moode using the CLI from a local repository:
```
$ git clone https://github.com/moode-player/moode.git
$ cd moode
$ git branch develop
$ git branch
  develop
* master
$ git checkout develop
Switched to branch 'develop'
$ git push
fatal: The current branch develop has no upstream branch.
To push the current branch and set the remote as upstream, use
    git push --set-upstream origin develop
$ git push --set-upstream origin develop
remote: Permission to moode-player/moode.git denied to pccasto.
fatal: unable to access 'https://github.com/moode-player/moode.git/': The requested URL returned error: 403
```

Successful attempt to create the develop branch in pccasto/moode using the CLI from a local repository:
```
$ git checkout -b develop
Switched to a new branch 'develop'
$ git push
fatal: The current branch develop has no upstream branch.
To push the current branch and set the remote as upstream, use
    git push --set-upstream origin develop
$ git push --set-upstream origin develop
Total 0 (delta 0), reused 0 (delta 0)
remote:
remote: Create a pull request for 'develop' on GitHub by visiting:
remote:      https://github.com/pccasto/moode/pull/new/develop
remote:
To https://github.com/pccasto/moode.git
 * [new branch]      develop -> develop
Branch 'develop' set up to track remote branch 'develop' from 'origin'.
```

moodeaudio, 2019-05-02:
I'm still learning Git admin but I did examine the access permissions and there are only two main choices (Owner or Member) that I could see for the people that are part of the moode-player org.

pccasto, 2019-05-03:
I'm not familiar with that aspect at all. I was surprised that I couldn't do it via the cli tools, but could using the web interface.  Not sure it matters yet, but wanted to capture the information in case it does, when we are creating branches to do pull requests.

pccasto, 2019-05-16:
Still don't know root cause, but I can (and did) push the 5.2+ changes to the develop branch using GitHub desktop, from my local repository.  Next time I attempt to do a push, I will capture some more information, and see if I can figure out why the cli does not work, and yet GitHub desktop does....

pccasto, 2019-05-16:
Reading the release notes for the git for window, it is remotely possible that the version I was running had a  bug around the handling of https authentication.  Seems odd that this would impact the moodeplayer/moode, and not pccasto/moode, but it is something worth investigating.  I've upgraded my version of git for windows, and will see what happens the next time I need to do a git push to develop branch.

moodeaudio, 2019-05-16:
Hi Paul,

Is there a Git GUI for Mac?

pccasto, 2019-05-16:
GitHub Desktop is available for MacOS. I use that on Windows, occasionally.  It seems to do many things, but I've tended to use other tools, so I don't fire it up very often.  Although looking at it again, maybe I could take more advantage of it.

Here's one list of some other GUI's that are available on MacOS.  Of these, I used GitKraken on Windows. Can't remember why I stopped.  That's the only one I'm familiar with. 
https://www.slant.co/topics/5574/~gui-git-clients-for-mac

Depending on what editor you use, it might have some GitHub integration. I use Atom, which is also available for MacOS, and it allows a lot of actions from within the editor. I haven't explored them all, because I've tended to use the command line for many things :-). But being able to run certain commands, and see the status of commits from within the editor is very handy.

pccasto, 2019-05-16:
And, in thinking about this question, I decided to check out VSCode - it is a cross-platform editor that has the same underlying engine as Atom, and has good git/GitHub integration.  From what I can see, I may replace Atom with VSCode.

You may be using Sublime - that seems to be popular on Macs -- it has a git integration as well.

moodeaudio, 2019-05-16:
I’ll check em out. I’m using Coda 2 which after checking seems to include a Git interface.

LeePorte, 2019-05-28:
Yes it'll be the permissions. You'll need to add everyone on the team `member` rights. otherwise forking the repo and creating the PR is the way to go for non team contributions.

closed 2019-06-13 by moodeaudio