moode-player/moode #32: Tidal and Qobuz integration

marcbth, 2019-05-24:
Hello,

it is possible to integrate tidal and qobuz into the surface. mpd 0.21xx already has the fundamental.
Then upnp would finally go away.
Alternatively an impletation, like volumio an upnp browser.
I'm currently working on providing moode for x86.

Many Thanks

Cliff

moodeaudio, 2019-05-24:
Hi Cliff,

Yes its possible to integrate Tidal and Qobuz into moOde but it will have to be done by another dev that has expertise in the API's. I can help do the config screens and music browser.

Can you clarify what you mean by UPnP browser? Is this the old djmount package?

marcbth, 2019-05-24:
At the moment I'm using upmpdcli and the asset upnp server. in upmpdcli I can activate tidal and qobuz. is solved via python.

volumio can then access it via an upnp browser.
but is not an ideal solution, because upnp. mpd sounds better.

from mpd 0.21x one can configure tidal and qobuz accesses.

Arthur from www.ompd.pl has recently tidal integration native. I can ask him or just look at git. the code could possibly take over.

Compliments to your work.

cliff

moodeaudio, 2019-05-24:
Cool thanks.

shujaaa, 2019-08-28:
Hello,

Any progress on that aspect ? I'm installing a new sound system and want to go with a solution that natively supports Qobuz/Tidal, and Moode is definitely a good looking/open source option !

Let me know !
Thanks,

closed 2019-09-04 by moodeaudio