moode-player/moode #34: Merge to master

moodeaudio, 2019-05-30:
Hey Paul,

I just released 5.3 and it was committed to the develop branch. Can you show me how to merge develop into master?

Then hopefully we can get to those PR's.

Thanks :-)

pccasto, 2019-06-02:
Hi Tim, 

Great!  I was traveling, so just saw this.  
This requires a couple of steps from the command line:
```
git checkout master
git merge --ff-only r53prod
```
You change into the branch where the merge will be made (master), and then do a fast forward merge based on the tag you have already created (r53prod).  (Note: this would have the correct results, even if there were additional merges on develop that had been made after the tag was created)

~paul

moodeaudio, 2019-06-02:
Do you have any experience with Github Desktop? 

I've been checking it out over the past few days and it seems to support the operations that otherwise would be done from cmd line.  For example I can switch to Master as current branch then merge Develop into Master. 

What I don't see though is any option to merge by tag, and I'm not sure whether the merge is automatically pushed to the Git repo or if I need to an addition step.

moodeaudio, 2019-06-02:
A separate question:

When I creates the r53prod tagged release I noticed that it referenced the sources in the Develop branch. When I merge Develop into Master do I need to create another tagged trelease based on Master?

pccasto, 2019-06-02:
You do need to do a push after the merge - I should have remembered to include that step.
Otherwise, you would have a merged copy, but the rest of us would not...

GitHub Desktop can do many of the things that can be done from the command line, but not all. It does not support all of the options. I think in this case (merging based on a tag), that the command line is the way to go.

The 'tagging' carries through.  With the fast-forward merge, you are doing an operation that makes a point on master that is the same as your tagged point on develop.  Tagging on develop (or even on a release branch), and then merging is the right approach.

moodeaudio, 2019-06-02:
Interesting.  So it's actually like below?

1. Commit the release to the Develop branch
2. Tag the release but don't publish it yet
3. Merge Develop into Master
4. Publish the tagged release

Also is there a way to configure the repos to automatically ignore the annoying OS X resource fork files?

The current workflow I use includes running a script that prunes out these hidden files before I do the add, commit,push.

pccasto, 2019-06-02:
Yes, that is a good order. The tag, merge, publish order makes sense.

Point #1 will get some refinement as we go along, because there will be commits to develop more frequently, but those commits don't always trigger the steps for releasing.  There are some steps that can be taken when multiple people are committing. There can be a 'release branch', made to make sure that the release has just what you want, and nothing else. But we don't need that yet...

You can (recommend you do) have a file at the top level of your moode repository called `.gitignore`. That file has pattern matches for files you don't want git to ever commit. That file itself should be committed, once you have some patterns in it.

moodeaudio, 2019-06-02:
I'll create the file.

Looking ahead, I'd like to start with your PR regarding CUE enhancements, then get to the PR from Chris that involves flattening the tag cache / library.

I'll need a download link to a zip containing some sample CUE files that can b used to test the enhancements.

pccasto, 2019-06-03:
I'll need to refactor my changes so that they commit against 5.3.  I'll have that done in a couple of days.

moodeaudio, 2019-06-04:
We should be all set wrt merging develop into master. I ended up doing it from GitHub Desktop cos its so easy. Also added in the .gitignore file.

moodeaudio, 2019-06-04:
Oops, out of force of habit I directly edited the readme.MD files in master and develop branches and ended up with a yellow warning indicating a PR needs to be made.

To try to recover I made the edits on my local copy and committed/pushed that to develop branch but yellow warning remains. 

Not sure how to fix this up...

pccasto, 2019-06-05:
I have reverted the two commits to master (in reverse order that the commits were made). So now master is back to the way it was when 5.3prod was released.  You can do this from either the command-line or GitHub Desktop. When you view history in you can right click a commit, and select revert, then push that to GitHub, and things are back to the way they almost were before.  (Almost, because the files are back to the way they were before, but because of the commits & reverts, the SHA fingerprint changes.)

When you did the merge from develop->master from GitHub, did you do it based on the tag? Is there an option for that once you select the branch to merge from?  I haven't tried that in GitHub Desktop, so I don't know if that is possible.  If there was an option for that, then that is perfect.  If there wasn't .... then It might have worked this time, because no other changes had been made to develop since the point where you declared that 5.3 was ready for production.  Some of the procedures for branching and tagging are to allow for parallel work. check-ins can be made that don't have any impact on a release that is being tested and released.

I'm not sure where the warning that a PR is required is coming from.  It looks like your direct commits to develop were successful.  What are you trying to do when you see that warning? From the settings, you can commit directly to develop, or master, without needing to do a PR. And from the commit history, it looks like you have been able to do that.  If you let me know where/when you are seeing that, I'll see if there is anything else I can figure out.

pccasto, 2019-06-05:
Perhaps what your are seeing is the yellow bar when comparing the develop branch with the master branch?  That has a button that allows you to create a pull request to merge develop into master, but does not require you to do so.   With this work-flow there wouldn't be a merge into master until you are ready to release.

pccasto, 2019-06-05:
I think I might have found what you were seeing.  A very recent update to GitHub Desktop (v 2.0.0) has this in the changelog: 

> Suggested next steps now includes suggestion to create a pull request after publishing a branch - [#7505](https://github.com/desktop/desktop/pull/7505)

Since we have designated `develop` as our main branch, I would think that the suggestion should not be offered.  Not sure if GitHub Desktop takes this into account, or not.  Also, the idea that every time you make changes to a branch that you are ready for a PR seems overly aggressive/optimistic. But I'll keep my eye on this, and see.

moodeaudio, 2019-06-05:
The yellow bar appeared when I viewed the repo directly via https://github.com/moode-player/moode. There is no yellow bar now :-)

Here is what I did using GitHub Desktop.

- Merge develop into master (there is no option to do it by tag)
- Edit /etc/shairport-sync.conf
- Edit /var/www/audioinfo.php
- Create .gitignore
- Commit / Push

Then I Edited readme.MD directly in develop and master repos. Thats when the yellow bar appeared.

moodeaudio, 2019-06-05:
What i was trying to achieve was to get .gitignore file and updated readme.md files in both develop and master but I see now that I broke the workflow by touching the master branch via commit and direct. It should only be  touched via a merge.

pccasto, 2019-06-05:
You can do most things from GitHub Desktop -- but when it comes time to release, I think it safest/best to use the CLI.  Since that is not very frequent, I hope this is OK with you. 

The day-to-day stuff - your own work, and pulling in/merging pull requests from others all should be easy within Desktop.

I think that yellow-bar is just advisory/suggestion, rather than a warning.

While you ...can... change files in two different branches, that sometimes causes conflicts. Another approach is to `cherry-pick` certain commits to merge from one branch to another.  But that is a more complex operation.  And it is not supported in GitHub desktop.

I created a small PR to use relative URLs in the README.md file -- fixing the underlying problem you were trying to solve with those commits to master and develop.  If you want to look at it, edit as you see fit, and merge that to develop, I can do a cherry-pick merge to master for you.

moodeaudio, 2019-06-05:
Yes of course we can use CLI for release.

I merged the PR into develop. Go ahead at your convenience and do the cherry pick  merge to master.

moodeaudio, 2019-06-05:
I must admit Git Desktop + Atom Editor is a powerful, easy to use combo.

moodeaudio, 2019-06-12:
Hi Paul,

Some minor issues in my second attempt at a release /merge. Master shows 5 commits ahead of develop after todays merge.

I'm not sure how to get them in sync.

chris-rudmin, 2019-06-12:
@moodeaudio Looks like you did commits to master which you never pulled back into develop.
https://github.com/moode-player/moode/network

Do a `git checkout develop` then `git merge master` then push it back to github.

moodeaudio, 2019-06-12:
Here's the relevant PR
https://github.com/moode-player/moode/pull/35

Let me have a look in Git Desktop.

moodeaudio, 2019-06-12:
Yes that worked perfectly. Thanks Chris :-)

closed 2019-06-13 by moodeaudio

pccasto, 2019-06-13:
Thanks for jumping in and answering @chris-rudmin.  Saw the email on this topic this morning. Took a look at the repository, and all looked correct, then looked at this thread and saw that all had been fixed.

chris-rudmin, 2019-06-14:
@pccasto @moodeaudio Since the 5.3.1 release, all my changes in #16 had merge conflicts (again) and it takes tons of time to go through and review each conflict. I think these conflicts could be avoided if you avoid using the features `git revert` and `git rebase` in your workflow which changes the git history. Thanks.

FYI you should only use `git rebase` if the commits in question have not been pushed the public repo. Also instead of `git revert`, just fix the error and make a new commit.  Doing this will make it much easier for 3rd parties to work in your code base.

moodeaudio, 2019-06-14:
I'm new to day-to-day use of Git but have not used rebase. I don't even know what it does. There were some reverts that paul made to fix up some of my noob mistakes but I think that's about it.