moode-player/moode #36: How to test PR's

moodeaudio, 2019-06-14:
I have the 4 PR branches #16, #18, #19, #29 loaded into Git Desktop but I'm not sure how to begin integrating and testing them against the 5.3.1 develop code base.

-Tim

chris-rudmin, 2019-06-14:
@moodeaudio Would you like to test them before you merge them into develop?
If so, you would need to checkout the remote branch. Otherwise you would merge the code, and then do your testing.

moodeaudio, 2019-06-14:
Depending on which branch is selected as the current branch in Git Desktop I see different versions of the source files under GitHub/moode dir tree. 

For example the Develop branch shows version 5.3.1 of playerlib.php while the PR16 branch shows 5.3 and the PR18 branch shows version 4.4.

I don't quite get what happens when PR18 with such old files gets merged into Develop.

chris-rudmin, 2019-06-14:
@moodeaudio If there is a conflict, it won't let you merge. If there is no conflicts then the code changes will be merged.  git commits are not atomic (with regards to the file), only the lines of code changed are part of the pull request.

moodeaudio, 2019-06-14:
Ok got it.

closed 2019-06-15 by moodeaudio