moode-player/moode #38: I2S audio device list don't have es9038q2m(or related ),but system seems have sabre driver

yaoyutian, 2019-06-19:
flac/mp3 music plays normal(select hifiberry dac/or other dac ...),but high bitrate encoder error,seems driver not  functioning!
my board is not Audiophonics,but chip is es9038q2m.
[image]
[image]
do i need recompile driver from here? https://github.com/audiophonics/I-Sabre_9038Q2M

yaoyutian, 2019-06-19:
after reboot from moode and shut down power ,reboot ,/boot/config.txt contents doesn't change;log info seems right,but still some file can't play (mpd error,fail to decode) .
[image]

yaoyutian, 2019-06-19:
find a modified version support DSD64...

moodeaudio, 2019-06-19:
moOde I2S driver list only supports devices that use drivers in stock Linux kernel. If your device does not work properly with any of the stock drivers, I cannot help.

closed 2019-06-20 by yaoyutian

yaoyutian, 2019-06-20:
I found this commit from moode forum:[Audiophonics I-Sabre 9038Q2M DAC driver ](https://github.com/raspberrypi/linux/commit/3d88199c2acde28f6b12d5e9a688a3c003c5e0e0)
And i found  i_sabre_q2m.ko in my burned moode SD system(5.3.1).
So i deemed moode 5.3.1 sould support my hardware play DSD64 .but turns out unable play.
But i searched a lot  forums ,got a modified 5.3.0 moode,burned it ,and it support DSD64.
I also found i_sabre_codec.ko in this system image.  so i guess hardware is ok, may some link error or other,i am trying found out .while my linux skill is at  a low level,i am still learning.
Close!

yaoyutian, 2019-06-25:
Raspbian buster desktop,directly modify /boot/config.txt :
dtparam=i2s=on
dtoverlay=i-sabre-q2m
vlc plays normally(flac,wav,DSD64) i2s sound out.