moode-player/moode #39: Change branch name

moodeaudio, 2019-07-04:
@pccasto,

I'd like to change branch name "develop" to "development"

Are steps below the right way to do this?

- note settings for develop branch
- create new branch development
- merge develop into development
- delete develop
- change settings on development to be same as develop

pccasto, 2019-07-19:
Hi Tim,

Sorry for the delay. We just finished a cross-country move (to AZ)  during which time my laptop decided to go unbootable, and present some challenges for data-rescue... The NAS with backups did travel in the car with us :-), but it still took a bit of doing to restore things. Everything's back working now.

Your approach seems correct. Would you like me to implement this at some point?

The 'develop' name for the branch seems to be a convention that many use - it was pretty well established as part of the GitFlow approach.. I'm not sure why that was chosen, rather than 'development' other than it is less typing... but I'm sure there were a number of debates. We are not bound by that convention, and can change it to whatever suits you best.

~paul

moodeaudio, 2019-07-19:
Moving is a lot of work! Enjoy your new digs  :-)

I guess what got me thinking about the branch name is that "develop" is a verb and "development" is a noun. In any case its low priority ATM cos of all the work around getting moOde 6 / Buster / Pi-4 shipped.

For the moment though I'm in a holding pattern because a couple of components are still a bit behind in their Buster updates for example WiringPi, USB boot, etc. 

I'm almost done with the moOde 6 features and updates so when u get the chance send me a download link to some test CUE/FLAC files that I can use to get going on your PR. I can probably figure out the mods from the code you posted so don't worry about fixing up the PR to be compatible with the current develop branch.

-Tim

ylesaout, 2019-09-26:
Hi,

I'm a user of moode since a few month now and i'm enjoying your work. I hope I will find time to contribute to the project.

By the way, if I had to change the branch name here is how I would do:
- First create the development branch on my computer 
```sh
$> git checkout develop # if you are not yet on develop branch
$> git pull # just to be sur to be up-to-date
$> git checkout -b development # here is the creation of the new branch which just is identical to the develop one (i.e. the new branch points at the same commit)
```
- Then push the new branch on github
```sh
$> git push --set-upstream origin development
```
- Copy github settings from develop branch to development branch
- Make some work on development branch and push it to github
- Check that everything work as expected (e.g. webhooks, continuous integration, ...)
- Delete the develop branch

Following these steps you can always go back if anything goes wrong before deleting the develop branch.

And note that you do not need to make any merge (either from develop to master or from develop to development). Indeed develop and development branch (before deleting develop branch) will be strictly identical.

moodeaudio, 2019-10-05:
Hi,

Thanks for the procedure :-)

closed 2024-10-18 by github-actions[bot]