moode-player/moode #40: Wifi password

Weasy666, 2019-07-09:
I have read your guide that states that quotes and double quotes are not allowed in the WiFi password. Is there a specific technical reason that this is not allowed? Because i have them in my WiFi password and this is the first time they are not allowed.

moodeaudio, 2019-07-10:
None. 

If you want to propose a fix then have a look at function cfgdb_update() in /var/www/inc/playerlib.php. I think what might work is to use SQLite3::escapeString() in place of the existing str_replace().

Also remove the pattern filter pattern="[^'\u0022]+" from /var/www/templates/net-config.html

If you are able to make these mods and verify that they work with your pwd that contains quotes then I'll modify on my end :-)

-Tim

Weasy666, 2019-07-10:
Ok...connected to my moOde per SSH and changed the 2 files with nano as you instructed. Afterwards the moOde connected to my WIFI without any problems. Looks like everything is working so far.

moodeaudio, 2019-07-10:
Thanks for testing. No need for PR, I'll make changes on my end. They will be included in moOde 6 :-)

closed 2019-07-10 by moodeaudio

moodeaudio, 2019-07-10:
After looking at the code it appears that the changes are a bit more complex than the test code I posted. 

If you re-save the network config be sure to type in the password again or the quotes will be lost.

You will not need to do this in moOde 6 :-)

Weasy666, 2019-07-11:
Ok, thanks. I never worked with PHP, so out of curiosity...are they automatically escaped from SQLite when loading from the database?

moodeaudio, 2019-07-11:
The string has to be escaped prior to storing it in the SQL table, then after retrieving it from the table and prior to displaying it in the HTML input control it has to be processed by the PHP function htmlentities() which turns double quote into &quot;. Single quotes can remain as-is.
https://www.php.net/manual/en/function.htmlentities.php

So in file /var/www/net-config.php the last line in the following code block (around line #139) should look like below.

```php
$_wlan0sec .= "<option value=\"wpa\"" . ($netcfg[1]['wlansec'] == 'wpa' ? 'selected' : '') . ">WPA/WPA2 Personal</option>\n";
$_wlan0sec .= "<option value=\"none\"" . ($netcfg[1]['wlansec'] == 'none' ? 'selected' : '') . ">No security</option>\n";
$_wlan0pwd = htmlentities($netcfg[1]['wlanpwd']);

```

Weasy666, 2019-07-11:
Ah...ok...now i get why you said re-save, i was thinking of a totally different place in code, not about the page rendering and html escaping. Thx for the info.

feamarth, 2020-01-23:
On Raspberry 3A+ I have still issues with wifi connection of moOde 6.4.0
No way to connect with an alphanumeric password setted.
Instead it connects if no password security is setted (so raspberry wifi is working).
Someone has some solution?