moode-player/moode #47: Show Genres column (Yes/No) on small display

llbstr, 2019-08-12:
Many thanks for version 6.0!!  👍 

I see an issue with library option _Show Genres column (Yes/No)_ on small display (my smartphone), see first screenshot for the bug, second screenshot is on landscape mode which goes without bug.

Getting same results with firefox or chrome (cleared cache completely), it happens if I'm in portrait mode after clicking on an artist. Went to 6.0 from fresh image.

[image]

[image]

moodeaudio, 2019-08-12:
I can repro. It's a bug and will be fixed in upcoming in-place update. 

As a workaround, just swipe to the right when in portrait and the Artist column will show up.

closed 2019-08-12 by moodeaudio

moodeaudio, 2019-08-12:
If you like you can try the code below. Add it to /var/www/css/moode.css right after line 251 in the media query: ```@media (max-width:479px) and (orientation:portrait)```. This comes compliments of @monkeypox :-) My tests were successful.

```
	/* for hide/show Genres column in Library Tag view */
	#top-columns.nogenre #lib-genre-header, #top-columns.nogenre #index-genres, #top-columns.nogenre #lib-genre {display:block;}
	#top-columns.nogenre #lib-artist-header, #top-columns.nogenre #lib-artist {left:50%;width:50%;}
	#top-columns.nogenre #index-artists {right:calc(0% + var(--sbw));}
	#top-columns.nogenre #lib-album-header, #top-columns.nogenre #lib-album {left:100%;width:50%;}
```