moode-player/moode #50: Python3-ready code

TheOldPresbyope, 2019-09-09:
This is not an immediate problem---indeed it may not raise its ugly head for some years yet---but the Python community is officially sunsetting Python2.7 on 2020/01/01. This means they will no longer provide security updates nor will they provide support to Python2.x coders, period.

Meanwhile, Debian included Python2.7 in Buster so it should at least run with its included libraries/modules until ~2022. I don't know what the official Raspbian position is but presumably it will be driven by Debian.

None of this comes as a surprise. Python 3.0 was released in 2008, but the user community has been dragging its feet in adopting it since the beginning. The official sunset date has been changed at least once as a result. There's more smoke than fire in the continual flame wars concerning 2 vs 3.

We can't boil the ocean but, as a practical matter, I think whatever Python code we put in moOde should be at least Python3 ready. 

A quick sweep of the current repo codebase turns up:
```
kreed@T520 ~/repos/moode $ find . -name \*py
./other/upmpdcli/session.py
./var/local/www/commandw/lcdup.py
./www/command/bt-agent.py
./www/command/gpio-buttons.py
```

I'm not sure we actually use session.py.

If I run pylint against the rest I get lots of messages relating to (non)conformance to PEPs, refactoring suggestions, etc., but only gpio-buttons.py is specifically flagged as not Python3 ready.

When I get a moment, I'll post some proposed fixes.

---

Notes:

- If one simply asks Raspbian Buster what depends on Python2.7, a laundry list of items pops up, the most important of which for us besides Python2.7 and its modules is Samba. I think as long as Samba is not up-versioned, it's safe to assume Python2.7 will stick around.

- Of course, these routines could be reimplemented in some other language :)

moodeaudio, 2019-09-09:
Thanks Kent :-)

almereyda, 2020-04-23:
This is more current now with the latest, last and final release of Python 2.7.18. As of now, there will be no future patches and/or releases.

TheOldPresbyope, 2022-01-03:
AFAIK RaspiOS bullseye does not include a python 2.x "in the box".

When I sweep moOde 7.6.1 /var directory for python modules I find the following 9 files

```
pi@rpi3bplcd:/ $ sudo find  /var -name \*py
/var/local/www/commandw/lcdup.py
/var/www/command/bt-agent.py
/var/www/command/upnp_albumart.py
/var/www/command/stationmanager.py
/var/www/command/mpdconfmerge.py
/var/www/command/pirev.py
/var/www/command/gpio-buttons.py
/var/www/command/camillaplotpipeline.py
/var/www/command/backupmanager.py
```
All but three of these explicitly invoke /usr/bin/python3

These three invoke /usr/bin/python instead
```
/var/local/www/commandw/lcdup.py
/var/www/command/bt-agent.py
/var/www/command/gpio-buttons.py
```
A quick test using the ast module shows that python3 can read all nine files and parse them into abstract syntax trees so I think we're in good shape. (scroll down to the second answer in [How to detect if code is python3 compatible](https://stackoverflow.com/questions/40886456/how-to-detect-if-code-is-python-3-compatible) to see example code.)

moodeaudio, 2022-01-03:
Right, Bullseye only has python3.
```
pi@moode8:~ $ python -V
Python 3.9.2
pi@moode8:~ $ python3 -V
Python 3.9.2
```

I'll update the three files.

bitkeeper, 2022-02-23:
I think we can close this issue, m8 is running exclusive on python 3  :-)

closed 2022-02-24 by TheOldPresbyope