moode-player/moode #53: Auto mounting XFS drives

matthijsvdmeulen, 2019-10-05:
As I've been experimenting with different filesystems for performance, I noticed that a USB drive formatted with the XFS filesystem does not automatically mount while a normal mount command does mount it correctly. As I've looked around, most solutions for auto mounting usb drives on raspbian lite mention some sort of script, does moode use this, or is there a particular package I need to nag about this issue?

moodeaudio, 2019-10-05:
Hi,

moOde 6 uses udisks-glue as the default auto-mounter but also offers udevil+devmon as an alternate auto-mounter cos udisks-glue segfaults on arm6. See System Config screen for changing the auto-mounter.

Udisks-glue supports many file systems. They are configured in /etc/udisks-glue.conf. Some info can be found at the link below.
http://manpages.ubuntu.com/manpages/precise/man5/udisks-glue.conf.5.html

Udisks-glue package is not available in Raspbian Buster repo so I built it by hand. See the moOde build recipe, STEP 3 - Install core packages,
https://github.com/moode-player/moode/blob/develop/other/build/build_recipe_v2.14.txt

Sources and other libs
https://github.com/moode-player/moode/tree/develop/other/udisks-glue

cd udisks-glue-master
./autogen.sh
./configure
make

udevil+devmon also supports many file systems but I'm not very familiar with its config. It seem to just work out-of-the -box. One type it apparently does not support are disk images. Udisks-glue does support mounting the partitions on disk images.

matthijsvdmeulen, 2019-10-06:
Thanks, I'll try and test my solution, and open a pr when I've tested that completely

closed 2019-10-30 by moodeaudio

themr0c, 2019-11-11:
Hi, I believe that the package `xfsprogs` also need to be installed.

moodeaudio, 2019-11-11:
Hi,

I have no way to test. Please ask the author of the PR to confirm whether the package is required. If it is I'll add it to upcoming moOde 6.4.0
https://github.com/moode-player/moode/pull/56

-Tim

matthijsvdmeulen, 2019-11-11:
Thanks for catching that, xfsprogs is indeed needed for xfs support in udisks-glue!

moodeaudio, 2019-11-11:
Great thanks for confirming :-)

PR #56 (2019-10-30): Add support for XFS filesystem to udisks-glue.conf
Fixes #53 
Adding the relevant config items has done the trick, I might look into udevil and devmon for support on the older hardware in the future, but this fixes what I need for now, and I hope other people find it useful as well.