moode-player/moode #55: replace mpd with mopidy

georgkrause, 2019-10-18:
Mopidy is a up to date re-implementation of mpd, and can be used as drop in replacement. Therefore I would suggest to replace mpd, in order to get some more features (like subsonic support)

moodeaudio, 2019-10-18:
It looks like there are functional gaps and limitations for example no support for crossfade, replay gain, output switching https://docs.mopidy.com/en/latest/ext/mpd/.

georgkrause, 2019-10-18:
Well, I feel like on the other side MPD also has a lot of missing features in contrast to mopidy, but the point not to remove features which users might need is valid.

Since there is already the option to choose a version of mpd, why not make mopidy also an option?

moodeaudio, 2019-10-18:
Sure, that's a good idea ;-) I'll add to the TODO list.

Have u actually rigged moOde to run mopidy?

georgkrause, 2019-10-18:
not yet, but I guess I can find some time to test this. I keep you updated

georgkrause, 2019-10-20:
It seems like the port of mpd (6600) is hard coded in several places, this makes it kind of hard to change this to 6680 for mopidy. 

Additionally something is restarting mpd right after I stop it. where can i disable this?

moodeaudio, 2019-10-20:
Have a look at /var/www/command/watchdog.sh

I did some quick searches through the code base for the port number and bin. See below.

```
# "6600"

/lib/systemd/system/mpd.socket
/etc/mpd.conf
/var/www/audioinfo.php
/var/www/chp-config.php
/var/www/command/index.php
/var/www/command/moode.php
/var/www/command/thmcache.php
/var/www/command/worker.php
/var/www/engine-mpd.php
/var/www/eqg-config.php
/var/www/eqp-config.php
/var/www/lib-config.php
/var/www/snd-config.php

# "/usr/local/bin/mpd"

/lib/systemd/system/mpd.service
/var/www/command/worker.php
```

georgkrause, 2019-10-20:
okay, this watchdog skript needs an update, port does not matter since mopidys mpd interface is also running on 6600, 6680 is an http interface.

I probably got it running, at least the webif does not show any errors. But adding streams does not work. need to test with some audio files later.

georgkrause, 2019-10-20:
file playback works, volume works. not working: add radio streams (playback with mpc works)

moodeaudio, 2019-10-21:
There prolly will be some breakage if the Mopidy protocol is not exactly = to the MPD protocol.

gerroon, 2019-11-06:
>Mopidy is a up to date re-implementation of mpd

No. Mopidy cant to multiple outputs like http, pulseaudio, fifo etc

moodeaudio, 2019-11-27:
Closing due to lack of activity. Feel free to reopen it if there is new information.

closed 2019-11-27 by moodeaudio