moode-player/moode #57: Qemu example?

gerroon, 2019-11-07:
Hi

Is it possible to run this on Qemu? If so any simple tutorial on how? I tried bunch of things and so far I could only get to the shell and login without any graphics. I am not sure if Qemu is emulaitng the sound hardware at the moment. In any case I would like to try this on Qemu before buying a pi to make sure I tested it first.

gerroon, 2019-11-07:
Here is the qemu running version

```
[  OK  ] Started Permit User Sessions.
ondemand
[  OK  ] Started OpenBSD Secure Shell server.
         Starting Disk Manager (legacy version)...
[  OK  ] Started Disk Manager (legacy version).
         Starting Authorization Manager...
[  OK  ] Started A high performanceΓÇªer and a reverse proxy server.
[  OK  ] Started The PHP 7.3 FastCGI Process Manager.
[  OK  ] Started Authorization Manager.
[  OK  ] Started /etc/rc.local Compatibility.
[  OK  ] Started Serial Getty on ttyAMA0.
[  OK  ] Started Getty on tty1.
[  OK  ] Reached target Login Prompts.

Raspbian GNU/Linux 10 moode ttyAMA0

moode login: pi\
Password:

Login incorrect
moode login: pi
Password:
Last login: Sun Oct  6 11:59:08 EDT 2019 on ttyAMA0
Kernel 4.19.75-v8+ #1270, Raspbian 10.1, moOde 6.3.0 2019-10-02
                          ____     __
              __ _  ___  / __ \___/ /__
             /  ' \/ _ \/ /_/ / _  / -_)
            /_/_/_/\___/\____/\_,_/\__/

                 moOde audio player
                (C) 2014 Tim Curtis

The programs included with Moode are free software; the
exact distribution terms for each program are described
in the individual files in /usr/share/doc/*/copyright.
Moode comes with absolutely no warranties expressed or
implied, or any other guarantees.

pi@moode:~$

```

gerroon, 2019-11-07:
This is the Qemu line

`qemu-system-aarch64 \
  -kernel bootpart_moode/kernel8.img \
  -initrd bootpart/initrd.img-4.14.0-3-arm64 \
  -dtb bootpart/bcm2837-rpi-3-b.dtb \
  -M raspi3 -m 1024 \
  -vga std \
  -serial stdio \
  -append "rw earlycon=pl011,0x3f201000 console=ttyAMA0 loglevel=1 vga=normal root=/dev/mmcblk0p2 fsck.repair=yes net.ifnames=0 rootwait memtest=1" \
  -drive file=moode-r630-iso.img,format=raw,if=sd \
  -net nic,model=rtl8139 -net user,hostfwd=tcp::20080-:80,hostfwd=tcp::20022-:22
`

moodeaudio, 2019-11-27:
Closing due to lack of activity. Feel free to reopen it if there is new information.

closed 2019-11-27 by moodeaudio