moode-player/moode #59: Responsive Web app

beankylla, 2019-11-10:
Hello,

Thanks for the awesome software I really love it!

I noticed something: on the moodeaudio website there is a possibility to have the website as an incon on mobile devices (progressive Web app). It basically makes an "app" from the website. You have it on your home screen and when you open it its in full screen without browser elements.

I found a link explaining what should be done. I believe the only important step. In our case is #1 :

https://appinventiv.com/blog/migrating-your-website-to-progressive-web-app/

It would be really awesome to have this for the moode player, it would save looking for the right tab in the browser part and would also partly answer to some people that were requesting an app.

I'm happy to help if I can tell me what I can do 😊.

moodeaudio, 2019-11-10:
Hi,

I don't quite understand the difference between PWA and simply "Add to Home Screen" which already makes an app (no Browser elements) out of http://moode, and it works on both Android and IOS.

Does it eliminate the manual steps involved in creating an app via "Add to Home Screen"?

beankylla, 2019-11-18:
The main added values are:
 it is a separate app than the browser. So when you browse your apps you can get straight to that one.
You can have it in fully full screen without seeing any of the browser  structure.

monkeypox, 2019-11-18:
I think Tim’s confusion is moode already is a progressive web app, and using the “add to home screen” command from the browser on iOS and Android already accomplishes what you’re asking for.

beankylla, 2019-11-20:
so for me it's there but only 50%.
As commented: a PWA is an app of its own. 

With the actual system, it opens up  in the browser. Benefit of this is that you can browse through apps and get straight it. Currently i have to browse through apps, and then find the right tab (or go to my home screen and click on the app)

On top you can remove any kind of browser overlay which makes the app fully full screen. Today you will always be in the browser which means that you have the browser overlay.

So no deal breaker here but i think this little extra would be really nice to have :)

eusi, 2019-11-25:
> As commented: once opened, a PWA is an app of its own.
> 
> With the actual system, it opens up the link in the browser.
Actually, on m Pixel 4 and Nexus 5x it works already as expected. "Add to Home screen" creates me an icon that is able to work standalone.
That means if I click on the icon it opens an own app-process (not within browser) in full screen (no address bar on top). 
I could provide screenshots of you want.
Maybe something is wrong on your phone's side?

moodeaudio, 2019-12-02:
I'll conclude from the discussion that "Add to Home Screen" is more or less equivalent to PWA although I'm not able to really understand @beankylla's explanation of PWA.

closed 2019-12-02 by moodeaudio

beankylla, 2020-04-04:
just adding a little comment: PWA seems to work exclusively on https.
I'm afraid a little downer.