moode-player/moode #60: Request : shairport-sync with MQTT client

zenman94, 2019-11-14:
Thank you for you great work on moode Audio.

Is it possible to provide a shairport-sync compiled with "--with-mqtt-client" in mOode Audio for a better integration in HomeAssistant that will be great!

moodeaudio, 2019-11-14:
I'll include in upcoming moOde 6.4.0 release :-)
```
pi@rp3:~ $ shairport-sync -V
3.3.5-OpenSSL-Avahi-ALSA-stdout-soxr-metadata-mqtt-dbus-sysconfdir:/etc
```

zenman94, 2019-11-14:
Nice !

closed 2019-11-20 by moodeaudio

janstadt, 2025-01-18:
I see the latest moode still has mqtt included. I've set the mqtt settings, but i am not seeing any updates on the topics i configured. 

```
mqtt =
{
	enabled = "yes"; // set this to yes to enable the mqtt-metadata-service
	hostname = "[ip]"; // Hostname of the MQTT Broker
	port = [port]; // Port on the MQTT Broker to connect to
	username = "[user]"; //set this to a string to your username in order to enable username authentication
	password = "[pwd]"; //set this to a string you your password in order to enable username & password authentication
//	capath = NULL; //set this to the folder with the CA-Certificates to be accepted for the server certificate. If not set, TLS is not used
//	cafile = NULL; //this may be used as an (exclusive) alternative to capath with a single file for all ca-certificates
//	certfile = NULL; //set this to a string to a user certificate to enable MQTT Client certificates. keyfile must also be set!
//	keyfile = NULL; //private key for MQTT Client authentication
	topic = "airplay/the-kitchen"; //MQTT topic where this instance of shairport-sync should publish. If not set, the general.name value is used.
//	publish_raw = "no"; //whether to publish all available metadata under the codes given in the 'metadata' docs.
//	publish_parsed = "no"; //whether to publish a small (but useful) subset of metadata under human-understandable topics
//	empty_payload_substitute = "--"; // MQTT messages with empty payloads often are invisible or have special significance to MQTT brokers and readers.
//    To avoid empty payload problems, the string here is used instead of any empty payload. Set it to the empty string -- "" -- to leave the payload empty.
//	Currently published topics:artist,album,title,genre,format,songalbum,volume,client_ip,
//	Additionally, messages at the topics play_start,play_end,play_flush,play_resume are published
	publish_cover = "yes"; //whether to publish the cover over mqtt in binary form. This may lead to a bit of load on the broker
//  enable_autodiscovery = "no"; //whether to publish an autodiscovery message to automatically appear in Home Assistant
//  autodiscovery_prefix = "homeassistant"; //string to prepend to autodiscovery topic
	enable_remote = "yes"; //whether to remote control via MQTT. RC is available under `topic`/remote.
//	Available commands are "command", "beginff", "beginrew", "mutetoggle", "nextitem", "previtem", "pause", "playpause", "play", "stop", "playresume", "shuffle_songs", "volumedown", "volumeup"
};

```

Is there anything else i need to do to get moode to publish status to mqtt?

janstadt, 2025-01-18:
I confirmed the client shows its connecting to my mqtt server, but nothing is being sent from moode/shairport-sync to the server. Is there something in those settings above that im missing so that it sends information to my mqtt server?

moodeaudio, 2025-01-18:
Questions and troubleshooting for a release are handled in the Support section of our Forum so that all our users can benefit from the discussion.

Visit the Support page at moodeaudio.org for information on how to easily request a Forum account.