moode-player/moode #61: NUL characters in text files

TheOldPresbyope, 2019-11-20:
I was surprised when I tried to view build_recipe_v2.15.txt using xed and got gibberish. Turns out it's considered a data file because the file contains a NUL character:

`kreed@T520 ~/repos/moode/other/build $ file build_recipe_v2.15.txt
build_recipe_v2.15.txt: data`

Running the following script prints the offending line and reveals its non-printing characters (NUL='^@' and LF='$'):
```
kreed@T520 ~/repos/moode/other/build $ grep -Pan '\x00' build_recipe_v2.15.txt|cat -vET
229:sudo apt-get -y install udevil^@$
```

I can't claim I've checked the entire repo, but I did look through ./www and its first-level subdirectories and found that www/players.php also contains a NUL character:
```
kreed@T520 ~/repos/moode/www $ grep -Pan '\x00' players.php|cat -vET
28:$result = shell_exec("avahi-browse -a -t -r -p | awk -F '[;.]' '/IPv4/ && /moOde/ && /audio/ && /player/ && /=/ {print $7\",\"$9\".\"$10\".\"$11\".\"$12}' | sort^@");$
``` 

I don't believe these NULs have done any harm in context but I believe they should be removed as a matter of good housekeeping.

moodeaudio, 2019-11-20:
Ok thanks. I don't know how those sneak in but I'll get rid of them in those two files.

moodeaudio, 2019-11-20:
Fixed

closed 2019-11-20 by moodeaudio