moode-player/moode #63: Electrical noise playing 16bit musics on ES9023 (Hifiberry DAC) board since ver.6.3.0

GoldJohnKing, 2019-11-26:
I've noticed obvious electrical noise since 6.3.0 update with my Pi2 + ES9023 board. Recently I've run a bunch of tests, and I'm pretty sure it's moode's problem.

| System | Source *bit | SoX to *bit | Noise 
| ------- | ------ | ------ | ------- |
moode | **16** | Off | **Yes**
moode | 24 | Off | No
moode | 32 | Off | No
moode | 24/32 | **Sox to 16** | **Yes**
moode | 16/32 | Sox to 24 | No
moode | 16/24 | Sox to 32 | No
volumio | 16/24/32 | Sox to 16/24/32 | No

Do you know what caused the probelm?

moodeaudio, 2019-11-26:
Prolly a bug in the dithering algorithm that MPD uses it its software volume control for 16 bit samples. You could try switching to hardware volume if your your DAC supports it or set SoX resampling to 32-bit / * kHz.

GoldJohnKing, 2019-11-26:
> Prolly a bug in the dithering algorithm that MPD uses it
I've switched to the experimental MPD, but it still happens. So both versions of the MPD have the problem? 

> its software volume control for 16 bit samples
It's sure caused by software volume control. I disabled software volume control and use 0db output, the noise does not occur anymore.

> switching to hardware volume
Hardware volume is not supported by my board.

> set SoX resampling to 32-bit / * kHz
Yes, it's what I do to avoid the problem.

So the problem is caused and would be fixed by MPD instead of moode?

GoldJohnKing, 2019-11-26:
It should be fixed by MPD in 0.22.

See:
https://github.com/MusicPlayerDaemon/MPD/issues/542
https://github.com/MusicPlayerDaemon/MPD/commit/af99f9fc9000e76990795a164dd51fd2bafc34c3
https://github.com/MusicPlayerDaemon/MPD/blob/af99f9fc9000e76990795a164dd51fd2bafc34c3/NEWS#L14

closed 2019-11-26 by GoldJohnKing