moode-player/moode #66: playerlib.php doesn't strip carriage returns from moodecfg.txt lines

TheOldPresbyope, 2019-12-06:
If a user creates a moodecfg.txt file with MSDOS line endings, e.g., '\r\n', function autoConfig copies the carriage returns as part of the parameter values. This results in what appears to the casual user to be a failed install.

Examining several obvious locations in the SD card of such a "failed" install, I can see the country code and the ssid in /etc/wpa_supplicant/wpa_supplicant.conf (not to mention the hostname in /etc/hostname) end in '\r', which means it wouldn't connect with the intended AP. I believe this may have been the case with user @kwatch in his recent post to the forum.

At a glance, it would seem that 

```
// Auto-configure settings at worker startup
function autoConfig($cfgfile) {
        autoCfgLog('autocfg: Auto-configure initiated');

        $contents = file_get_contents($cfgfile);
        $autocfg = array();
        $line = strtok($contents, "\n");
```

might use instead something like

`$line = str_replace(array("\n", "\r"), '', $contents);`

but I'm no PHP guru, just a hacker who knows how to cherrypick Google results:)

moodeaudio, 2019-12-06:
Nice catch. 

Prolly change
```
$contents = file_get_contents($cfgfile);
```
to

```
$contents = str_replace("\r\n", "\n", file_get_contents($cfgfile));
```

moodeaudio, 2019-12-07:
Fix confirmed by @TheOldPresbyope :-)

closed 2019-12-07 by moodeaudio