moode-player/moode #67: UI rewrite

ekozan, 2019-12-10:
Hi 

I dont know if this idea will be loved :D 

But I start to rewrite all the UI to Angular, why for fun and for more modularity 
and code cleaning 

I can show you an POF in 2 week or less i already got the player working great 

My idea is to upgrade adaptative option for feat to smartphone and 7" 

If you love that i'll try to port every curent option to angular 

Max,

moodeaudio, 2019-12-10:
Hi Max,

I'm always interested in improving moOde but the challenge is that no one on the moOde project has any experience with Angular.

-Tim

monkeypox, 2019-12-10:
Hi Maxine,

It won’t be unloved. :) It might be easier to aim for “moode next” rather than try to recreate absolutely everything. Talk to Tim 1-on-1 and see what features are essential to moode and work towards that.

I have some in-progress ui I can send as your project progresses so you can skate to where the puck is headed, I don’t think image attachments work in email on a list like this but I can upload to the git thread directly when I have time.

monkeypox, 2019-12-10:
Here’s one of the main changes - switching to a menu for the different views rather than the button bar which lets us use the middle of the screen for descriptive text.

[image]

Search is in the menu instead (keyboard cropped):

[image]

Modals no longer have window chrome (large screens it’s centered rather than full screen)

[image]

Backdrop options are hidden when alpha = 1 (they wouldn’t show) and as above only the choose or remove button is shown at any one time.

[image]

I’ll upload a branch tonight if you want to see it in action though some of the things in there won’t necessarily make it to moode, principally the alternate large screen layout which doubles as a coverview rethink.

ekozan, 2019-12-12:
I take the idea :D 

For the momment i make the base 

For memory you can follow my work here : https://github.com/ekozan/moode-ui
for the moment no many thing to see ( just a player bar not finished  )

ekozan, 2019-12-13:
Idea for dont forget and discution : 

Hability to configure action click or touch on something 

Type of click available : 
- Click 
- DblClick
- ContextClick
- Touch 
- HoldTouch 
- Keybord key ? :) ( space, arrows ) 

Action : 
- Play
- Next
- Prev
- Pause
- Open ( info, album in tracks windows etc... )
- Open context menu 
- Add to playlist

inta, 2020-01-01:
Are you sure Angular is the right choice here? The current web interface based on Bootstrap and jQuery is not the fastest on weak devices and I think Angular will add some more weight on top. I would choose a smaller framework like Mithril, Preact or – if you really need a famous name – Vue. Angular has a really steep learning curve that only pays off for large projects with large teams.

Just my 2 cents as a moode audio user with a web development background.

moodeaudio, 2020-02-26:
Closed due to lack of activity.

closed 2020-02-26 by moodeaudio