moode-player/moode #69: Please prevent enabling Bluetooth without enabling WiFi

sebastiendu, 2019-12-14:
With 6.4.0 running on Raspberry Pi 3B+ (not tested other configurations),

in order to enable _Audio Config_ / _Renderers_ / _Bluetooth_
enabling _System Config_ / _System Modifications_ / _Integrated BT adapter_ is not enough.
One needs to also enable _Integrated WiFi adapter_.

Without this, trying to enable Bluetooth causes a web interface loss.

The only way out I found is
`sqlite3 /var/local/www/db/moode-sqlite3.db "update cfg_system set value=0 where param='btsvc';" && reboot`

Is it possible to create dependency links between these three options ?

moodeaudio, 2019-12-14:
Quick test on Ethernet connected 3B+ and I'm not able to repro. 

Below is after turning off Integrated WiFi adapter in System Config then rebooting

```
pi@moode:~ $ cat /boot/config.txt
disable_splash=1
disable_overscan=1
hdmi_drive=2
hdmi_blanking=1
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=audio=on
dtoverlay=pi3-disable-wifi
#dtoverlay=pi3-disable-bt

pi@moode:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.181  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::7530:44ab:489f:717f  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:ec:46:b2  txqueuelen 1000  (Ethernet)
        RX packets 311  bytes 41473 (40.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 243  bytes 66822 (65.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 113  bytes 11752 (11.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 113  bytes 11752 (11.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@moode:~ $ moodeutl -t
3847 worker: MPD output 5 ALSA polarity inversion (off)
20191214 123847 worker: MPD output 6 ALSA bluetooth (off)
20191214 123847 worker: MPD output 7 HTTP stream (off)
20191214 123847 worker: MPD crossfade (off)
20191214 123847 worker: Audio source (Local)
20191214 123847 worker: Output device (Local)
20191214 123847 worker: USB auto-mounter (udisks-glue)
20191214 123847 worker: -- Music sources
20191214 123847 worker: USB sources (none attached)
20191214 123847 worker: NAS and UPnP sources (none configured)
20191214 123847 worker: -- Miscellaneous
20191214 123847 worker: Saved MPD vol level (0)
20191214 123847 worker: Preamp volume level (0)
20191214 123847 worker: MPD volume level (0) restored
20191214 123847 worker: ALSA volume level (100%)
20191214 123847 worker: Auto-play (Off)
20191214 123847 worker: Maintenance interval (6 hrs)
20191214 123847 worker: Screen saver activation (Never)
20191214 123847 worker: Session permissions (OK)
20191214 123847 worker: Watchdog started
20191214 123847 worker: Ready
20191214 124027 worker: Job btsvc
20191214 124035 worker: Bluetooth controller initialized
```

On my iPhone, "Moode Bluetooth" appears in list of available Bluetooth devices.

moodeaudio, 2019-12-15:
Feel free to reopen if u can provide a repro.

sebastiendu, 2019-12-15:
My setup is a little different, using `dtoverlay=hifiberry-dacplus` and therefore `#dtparam=audio=off`,
which might cause the problem.

```
pi@entree:~ $ cat /boot/config.txt 
disable_splash=1
disable_overscan=1
hdmi_drive=2
dtparam=i2c_arm=on
dtparam=i2s=on
#dtparam=audio=off
gpu_mem=16
dtoverlay=hifiberry-dacplus
dtoverlay=pi3-disable-wifi
#dtoverlay=pi3-disable-bt
pi@entree:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.18  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::689:8751:f140:51e1  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:c2:d5:ce  txqueuelen 1000  (Ethernet)
        RX packets 84  bytes 12216 (11.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 196  bytes 47018 (45.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 349  bytes 36944 (36.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 349  bytes 36944 (36.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@entree:~ $ moodeutl -t
ossfade (off)
20191215 090055 worker: Audio source (Local)
20191215 090055 worker: Output device (Local)
20191215 090055 worker: Airplay receiver started
20191215 090055 worker: UPnP renderer started
20191215 090055 worker: DLNA server started
20191215 090055 worker: USB auto-mounter (udisks-glue)
20191215 090055 worker: -- Music sources
20191215 090055 worker: USB sources (none attached)
20191215 090056 worker: NAS and UPnP sources (mountall initiated)
20191215 090056 worker: -- Miscellaneous
20191215 090056 worker: Saved MPD vol level (0)
20191215 090056 worker: Preamp volume level (0)
20191215 090056 worker: MPD volume level (34) restored
20191215 090056 worker: ALSA volume level (100%)
20191215 090056 worker: Auto-play (Off)
20191215 090056 worker: Maintenance interval (6 hrs)
20191215 090056 worker: Screen saver activation (Never)
20191215 090056 worker: Session permissions (OK)
20191215 090056 worker: Watchdog started
20191215 090056 worker: Ready
20191215 090057 watchdog: INFO Audio output is (closed)
20191215 090328 worker: Job btsvc
```
("Bluetooth controller initialized" never appears)

```
pi@entree:~ $ tail -n 15 /var/log/syslog
Dec 15 09:01:19 entree systemd-timesyncd[247]: Synchronized to time server for the first time 212.83.154.33:123 (2.debian.pool.ntp.org).
Dec 15 09:03:29 entree systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Dec 15 09:03:29 entree kernel: [  166.756003] uart-pl011 3f201000.serial: no DMA platform data
Dec 15 09:03:59 entree btuart[1672]: Initialization timed out.
Dec 15 09:03:59 entree btuart[1672]: bcm43xx_init
Dec 15 09:03:59 entree systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Dec 15 09:03:59 entree systemd[1]: hciuart.service: Failed with result 'exit-code'.
Dec 15 09:03:59 entree systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Dec 15 09:03:59 entree systemd[1]: Condition check resulted in Bluetooth service being skipped.
Dec 15 09:03:59 entree systemd[1]: Condition check resulted in Bluetooth service being skipped.
Dec 15 09:03:59 entree systemd[1]: Started BluezAlsa proxy.
Dec 15 09:04:00 entree dbus-daemon[302]: [system] Activating via systemd: service name='org.bluez' unit='dbus-org.bluez.service' requested by ':1.16' (uid=0 pid=1763 comm="/usr/bin/bluealsa -p a2dp-source -p a2dp-sink ")
Dec 15 09:04:00 entree dbus-daemon[302]: [system] Activation via systemd failed for unit 'dbus-org.bluez.service': Unit dbus-org.bluez.service not found.
Dec 15 09:04:00 entree bluealsa[1763]: /usr/bin/bluealsa: Couldn't get managed objects: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit dbus-org.bluez.service not found.
Dec 15 09:11:39 entree systemd[1]: Started Session c2 of user pi.
```

sebastiendu, 2019-12-15:
> Feel free to reopen if u can provide a repro.

I cannot re-open my own issue because you (as a repo collaborator) closed it. Github feature.

moodeaudio, 2019-12-15:
Gotcha.

moodeaudio, 2020-02-22:
Hi,

I was able to run a quick test and am still not able to repro...

```
Authenticated to rp2 ([192.168.1.228]:22).
Kernel 4.19.102-v7+ #1295, Raspbian 10.3, moOde 6.5.0 2020-MM-DD
                          ____     __   
              __ _  ___  / __ \___/ /__ 
             /  ' \/ _ \/ /_/ / _  / -_)
            /_/_/_/\___/\____/\_,_/\__/ 
                                          
                 moOde audio player 
                (C) 2014 Tim Curtis

The programs included with Moode are free software; the
exact distribution terms for each program are described
in the individual files in /usr/share/doc/*/copyright. 
Moode comes with absolutely no warranties expressed or 
implied, or any other guarantees.

Last login: Sat Feb 22 08:29:36 2020 from 192.168.1.125
pi@rp2:~ $ cat /boot/config.txt 
disable_splash=1
disable_overscan=1
hdmi_drive=2
hdmi_blanking=1
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=audio=off
dtoverlay=hifiberry-dacplusadc
dtoverlay=pi3-disable-wifi
#dtoverlay=pi3-disable-bt
pi@rp2:~ $ 
pi@rp2:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.228  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::8bad:d5a5:ff01:16b2  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:81:0a:bf  txqueuelen 1000  (Ethernet)
        RX packets 695  bytes 280262 (273.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 131  bytes 19943 (19.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 83  bytes 8911 (8.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 83  bytes 8911 (8.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pi@rp2:~ $ 
pi@rp2:~ $ moodeutl -t
ource (Local)
20200222 083600 worker: Output device (Local)
20200222 083600 worker: Bluetooth pairing agent started
20200222 083600 worker: USB volume knob (Off)
20200222 083600 worker: USB auto-mounter (udisks-glue)
20200222 083600 worker: -- Music sources
20200222 083600 worker: USB sources (none attached)
20200222 083600 worker: NAS and UPnP sources (none configured)
20200222 083600 worker: -- Miscellaneous
20200222 083600 worker: LED0 trigger (mmc0)
20200222 083600 worker: LED1 brightness (255)
20200222 083601 worker: Saved MPD vol level (0)
20200222 083601 worker: Preamp volume level (0)
20200222 083601 worker: MPD volume level (14) restored
20200222 083601 worker: ALSA volume level (100%)
20200222 083601 worker: Auto-play (Off)
20200222 083601 worker: Auto-shuffle service (Off)
20200222 083601 worker: Maintenance interval (3 hrs)
20200222 083601 worker: Screen saver activation (Never)
20200222 083601 worker: Session permissions (OK)
20200222 083601 worker: Watchdog started
20200222 083601 worker: Ready
20200222 083908 worker: Job btsvc
20200222 083916 worker: Bluetooth controller initialized

```

moodeaudio, 2020-02-26:
Closed due to inability to repro.

closed 2020-02-26 by moodeaudio

berwegera, 2020-04-25:
I can reproduce this with a hifiberry amp2 on a raspberry pi 3 Pi 3 Model B Plus Rev 1.3
with Moode 6.4.2 and moode 6.5.0
But the error is gone with hifiberry amp2 (same as before) and Raspberry Pi 4 Model B Rev 1.1 with moode 6.4.2 and 6.5.0

moodeaudio, 2020-04-25:
I don't have an Amp2 board so there is no way for me to try and repro.

berwegera, 2020-04-27:
No Problem. For me the problem is pi3 specific and so i use a pi4 (there is no problem)  with the hifiberry amp2

nkasrin, 2020-07-31:
I too have faced the same problem reported by @sebastiendu and the solution he proposed worked for me.
My setup, Khadas Toneboard usb dac and:

Release: 6.7.1 2020-07-22

RaspiOS: 10.4
Linux kernel: 5.4.51-v7+ #1325
Platform: Pi-3B+ 1GB v1.3
Architecture: armv7l
MPD version: 0.21.24

gebla, 2021-01-09:
Using 3B+ with Hifiberry digi+ I/O hat i have the same issue. With Wifi disabled and BT enabled in system settings I loose webinterface. With Wifi and BT enabled BT is working and webinterface is working as well.

System settings: 
Release Information
Release: 7.0.1 2020-12-23
Coding: Tim Curtis © 2014
Documentation: View release notes, View setup guide
Contributors: View contributors
License: View GPLv3
Platform Information
RaspiOS: 10.6
Linux kernel: 5.4.77-v7+ #1371
Platform: Pi-3B+ 1GB v1.3
Architecture: armv7l
MPD version: 0.22.3_p0x3

moodeaudio, 2021-01-09:
Quick test on a 3B+ and no issues on my end with having just the Integrated WiFi adapter turned off. Ethernet connection works perfectly.

gkby79, 2023-12-10:
Same behaviour for me after an update last night. Raspberry pi zero, had an amp shim which uses snd_rpi_hifiberry_dac, and if Bluetooth is on, it blocks the web ui with an error about Bluetooth controls being on

ppdms, 2024-01-12:
This happened to my zero 2w, but turning off wifi from settings is enough to fix it

gkby79, 2024-01-12:
My use case is connected to WiFi for streaming Spotify, and occasionally use the web ui to take back control and switch to radio, so turning WiFi off isn't an option for me

moodeaudio, 2024-01-12:
Your best bet is to post the issue in the support section of. our Forum so someone can try to reproduce it. The Support page at moodeaudio.org has information on how to easily request a Forum account.