moode-player/moode #71: No static IP address, because eth1 is used for external network dongle

sphh, 2020-01-19:
I have a LTE (mobile internet) dongle plugged into the RPi 1B to provide the internet address. I configured a static ethernet address through the web interface, but the RPi never acquired it. It always got the address from the DHCP server on the LTE dongle. After I figured out, what the web interface does, I noticed that the `/etc/dhcpcd.config` contains `interface eth0`! After changing this line to `interface eth1`, everything works as expected (until the file gets overwritten by a new save).

I would think, that effects everyone who uses a USB dongle showing up as `eth1` (because the internal network interface is `eth0`). Would it be possible to write a configuration file which will also work for USB network dongles?

My system:
moOde release		= 6.4.0 2019-11-24
Raspbian OS		= 10.2

moodeaudio, 2020-01-19:
That could be difficult cos the code is designed around a single active network interface, either eth0 or wlan0 and this works well for the vast majority of users. Have a look at worker.php, net-config.php and the cfg_network table to see where the persistent names eth0 and wlan0 are hard-coded.

Maybe there is some sort of kernel hack or command that can make eth0 an alias for eth1. Something like that,

sphh, 2020-01-19:
I have two or three ideas, how that may be done:

1. The web interface simply shows two ethernet interfaces. Easy, but not beautiful. Maybe the configurations for the second interface `eta1` is only shown, if it can be found, reducing clutter, but still makes it possible to configure it.
2. During loading time the web interface checks if there is an active interface. If it finds one, it configures this interface, otherwise it configures `eth0`.
3. Make a drop down box with the two possible interfaces `eth0` to select the interface.

Personally I would prefer 1.

moodeaudio, 2020-02-26:
Closed due to moOde being designed around a single active network interface, either eth0 or wlan0.

closed 2020-02-26 by moodeaudio