moode-player/moode #73: Favorites.m3u formatted incorrectly

duracell80, 2020-01-29:
Spotted on this thread
http://moodeaudio.org/forum/showthread.php?tid=1925

If a user is playing a station that's not in the radio database table and then adds the currently playing station to their favorites by way of clicking the heart icon in the player ... the resulting Favorites.m3u file is just a list of URL's without associated "track titles".

This then makes it harder to navigate the Favorites playlist.

[image]

[image]

[image]

moodeaudio, 2020-01-29:
Regarding M3U formatting, the MPD command "save" is used. This command only writes the URL to the file because there is no metadata stored in MPD queue. It just contains URL's or file paths.

The issue where the station name from cfg_radio is not being displayed after add/edit a station or adding to favs is likely the result of RADIO.json array in playerlib.js not being updated. A page refresh will update the array but we want to make it so this is not needed. I'll have a look.

monkeypox, 2020-01-29:
EXTM3U is actually an extension to M3U, see:

http://www.assistanttools.com/articles/m3u_playlist_format.shtml

We talked about this a little before. It’d be nice to support a more full featured format (.pls would also work and is what the radio stations currently use iirc) but as Tim said this is what mpd currently supports and doing something different requires someone to actually do it. :)

moodeaudio, 2020-02-01:
After looking at the code it appears that indeed the session is not being updated when a station is deleted. This has caused an accumulation of deleted stations in the session file over time.

The fix will be included in 6.4.2 update.

moodeaudio, 2020-02-26:
Closed. Fix was provided in 6.4.2

closed 2020-02-26 by moodeaudio