moode-player/moode #78: Library work items

moodeaudio, 2020-02-28:
Hi,

PR 76/77 from @gbh-uk tested and merged into the develop branch :-)

@monkeypox, doesn't this more or less cover the Album/Year sort when Artist clicked scenario?

Just a heads up regarding some items that I'm looking at working on to further improve the Library for r650 release. These are relatively easy for me to do on my own.

1. Use "sort" instead of "group" in the code. This involves changing library_album_grouping to library_album_sort in cfg_system and in 6 source files.
2. Migrate Library options edit from PHP lop-config to JS Appearance dialog. I'll prolly need a bit of @monkeypox code that does the heading expand/collapse.
3. Add an appearance option to turn ON/OFF Tag view covers
4. Test an appearance option to turn ON/OFF ellipsis-limited lines for Album view

A more complex improvement that I'll need help to implement is @monkeypox changes that replace the button bar in Library views with a more flexible Menu. There are also IIRC a few improvements to the Playbar that should be implemented.

monkeypox, 2020-02-28:
I try to get some PRs with some of my changes done soon. I'll just do the minor, non-controversial ones for now.

Album>Year was for the case where the album column was sorted by album and a selected artist sorted the resulting albums chronologically.

We should have separate tag & cover view preferences, that's pretty simple though.

moodeaudio, 2020-02-28:
Thats what I thought re Album/Year.

monkeypox, 2020-02-28:
Gary this convo is in “Issues”.

The album column in tag view shouldn’t necessarily be the same thing as the cover view list of albums. For example if you wanted artist sorting in cover view but album sorting in the tag view album column. Anything but album sorting in the album column is actually kind of an outlier as moode is the only player I know of that allows it (or album covers).

moodeaudio, 2020-02-28:
I actually find the "by Artist" or "by Artist/Year" sort in the Albums column pretty useful cos I normally try to find things (click, search or quick index) by Artist rather than Album name.

moodeaudio, 2020-03-04:
Number 1-4 are done and committed :-)

closed 2020-03-15 by moodeaudio