moode-player/moode #93: Provide PINN image

phobos-dev, 2020-03-25:
Hi,

would it be possible to provide an PINN image for integration in that project. This way one could easily install moOde in a multi OS environment. Also it would help to get the project greater and widely spread around the globe. Would be great.

Thanks for considering.

References:
https://github.com/procount/pinn
procount/pinn#358

moodeaudio, 2020-03-26:
Not possible. moOde it tightly integrated into it's own custom version of Raspbian Lite called MoodeOS.

closed 2020-03-26 by moodeaudio

procount, 2022-01-16:
MoOde v7.6.1 has now been converted for installation using PINN.
If you are interested in maintaining PINN images for future versions of moOde, please get in touch.

lurch, 2022-01-25:
Just to clarify, PINN is an OS-installer (so you'd use it to install MoodeOS), it's not an application installer :slightly_smiling_face: