moode-player/moode #95: Hardware volume control and devices with mic ins

lych, 2020-03-27:
Hi!
I've been trying Behringer UMC204HD.
The device has two microphone inputs and two outputs.
```
root@moode:/etc# amixer -c 1
Simple mixer control 'Mic',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 127
  Front Left: Capture 127 [100%] [0.00dB] [on]
  Front Right: Capture 127 [100%] [0.00dB] [on]
Simple mixer control 'Mic',1
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 127
  Mono: Capture 127 [100%] [0.00dB] [on]
Simple mixer control 'UMC204HD 192k Output',0
  Capabilities: pvolume pswitch
  Playback channels: Front Left - Front Right - Rear Left - Rear Right
  Limits: Playback 0 - 127
  Mono:
  Front Left: Playback 127 [100%] [0.00dB] [on]
  Front Right: Playback 127 [100%] [0.00dB] [on]
  Rear Left: Playback 127 [100%] [0.00dB] [on]
  Rear Right: Playback 127 [100%] [0.00dB] [on]
Simple mixer control 'UMC204HD 192k Output',1
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 127
  Mono: Playback 127 [100%] [0.00dB] [on]
```
And
`awk -F"'" '/Simple mixer control/{print "(" $2 ")";}' <(amixer -c $CARD_NUM)`
returns
```
(Mic)
(Mic)
(UMC204HD 192k Output)
(UMC204HD 192k Output)
```
It's not exactly the desired result and hardware volume control doesn't work (probably it works, but
controls a wrong mic channel).

I think, checking the capabilities section of mixers will help.
Somehow like this:
`amixer -c $CARD_NUM | awk 'BEGIN{FS="\n"; RS="Simple mixer control"} $0 ~ "pvolume" {print $1}' | awk  -F"'" '{print "(" $2 ")";}'
`
So, in my case the result is
```
(UMC204HD 192k Output)
(UMC204HD 192k Output)

```

moodeaudio, 2020-03-27:
Thanks. It looks like a nice improvement :-) I'll test it out for upcoming r650 release.

moodeaudio, 2020-04-24:
Implemented.

closed 2020-04-24 by moodeaudio