moode-player/moode #96: MPD only bound to IPv4 interface

Finomnis, 2020-03-29:
It seems that the MPD config of Moode does not support IPv6.
This is a shame in an IPv4/IPv6 mixed network, as done with modern fritzbox routers. That means you have to manually enter the IPv4 in every MPDroid client instead of performing a dns lookup.

Info:
moOde 6.4.2 2020-02-12
Raspberry Pi 3 B+
Network: 5G Wifi

MPDroid Error Message:
"failed to connect to <dns-name>/<ipv6> (port 6600) from /<ipv6-phone> (port 34743) after 10000ms: isConnected failed: ECONNREFUSED (Connection Refused)"
 - MPDroid Works with IPv4

moodeaudio, 2020-03-29:
Why is this a moOde issue? Our project does not support any 3rd party MPD clients.

Finomnis, 2020-03-29:
It doesn't? Oh, didn't realize that. Is there any reason for that? Because port 6600 is open and third party clients seem to work ...

moodeaudio, 2020-03-29:
The dev team only provides support and troubleshooting for moOde's built-in WebUI. For 3rd party MPD clients it's the maintainers responsibility if it doesn't work with moOde. Users that use 3rd party MPD clients certainly do help each other out in our Forum so you may want to post your issue there and see if any MPDroid users can help out :-)

Finomnis, 2020-03-29:
Thanks. I don't think it's anything the 3rd party clients can fix, but after more research, I also don't think moOde could fix it. It seems like MPD itself is unable to listen on both ipv4 and ipv6 adresses simultaneously: https://wiki.archlinux.org/index.php/Music_Player_Daemon/Troubleshooting#Binding_to_IPV6_before_IPV4

Thanks and thank you for all your work in this project! Highly appreciated.

closed 2020-04-03 by Finomnis