moode-player/moode #99: Sound distorted for open.fm radio

michalwu, 2020-04-09:
Hi

Latest moOde Player. Audio for specific web radio is silent and garbled. I have had same problem with volumio. There is no audio problem when I open same stream via browser or android app. 
From what I was able to get from tech people working there, they are using AAC+ (HE-AAC v2).

I already have solution: Just edit /etc/mpd.conf and add:
decoder {
plugin "faad"
enabled "no"
}

Any chance to fix this permanently? Now I need to paste this from time to time as something is restoring original file.

Source: | http://stream.open.fm/20
-- | --
Encoded at: | VBR compression
Decoded to: | 24 bit, 44.1 kHz, Stereo, 46 kbps
Destination: | Local
Output rate: | 24 bit, 44.1 kHz, Stereo, 2.117 Mbps

Volume ctl: | Software (MPD 32-bit float with dither)
-- | --
Resampling: | off
Polarity inv: | off
Crossfade: | off
Crossfeed: | off
Parametric EQ: | off
Graphic EQ: | Classic V Plus
Replaygain: | off
Normalize vol: | off
Chip options: | None

Device: | HiFiBerry Digi(Digi+)
-- | --
Chip: | Cirrus Logic WM8804
Interface: | I2S
Formats: | S16_LE, S24_LE
Platform: | Pi-3B 1GB v1.2

moodeaudio, 2020-04-10:
I'm not having any issues playing the station.
```
INPUT / OUTPUT
Source: 	http://stream.open.fm/20
Encoded at: 	VBR compression
Decoded to: 	16 bit, 44.1 kHz, Stereo, 44 kbps
Destination: 	Local
Output rate: 	32 bit, 44.1 kHz, Stereo, 2.822 Mbps

DSP OPERATIONS
Volume ctl: 	Hardware (on-chip volume controller)
Resampling: 	off
Polarity inv: 	off
Crossfade: 	off
Crossfeed: 	off
Parametric EQ: 	off
Graphic EQ: 	Classic V
Replaygain: 	off
Normalize vol: 	off
Chip options: 	Apodizing Fast Roll-off Filter, De-emphasis=Bypass, DoP=on

AUDIO DEVICE
Device: 	Allo Katana DAC
Chip:	 	ESS Sabre ES9038Q2M
Interface: 	I2S
Formats: 	S16_LE, S32_LE
Platform: 	Allo USBridge SIG [CM3+ Lite 1GB v1.0]
```

michalwu, 2020-04-10:
HW difference then? What can I do?

moodeaudio, 2020-04-10:
You could try setting SoX resampling to 32 bit / * kHz in MPD config to see if that might fix your issue but since Im not able to reproduce it I can't offer any further help.

moodeaudio, 2020-04-10:
That should be 16 bit / * kHz since your audio device reports that it only supports 16 and 24 bit.

closed 2020-04-24 by moodeaudio