moode-player/moode #100: Feature Request: Folder support in Playlist Folder

duracell80, 2020-04-25:
A quick demo of adding a folder and adding a playlist into the folder.
https://youtu.be/cGhYCdaPMfY

Can it be possible to add folders in the playlist share as a "Music Source" in the Library configuration. Currently we can add SMB shares and NAS drives which is cool and I suppose if we had local share on the pi that could be a "source" too. And I suppose something could be done with the SDCARD share where a playlist folder could live there. But yeah currently you can't add folders to the mpd playlist directory and have them in the UI.

I can do it on the SDCARD for sure, but it would be nice to say have a Podcast folder right there in playlists.

monkeypox, 2020-04-25:
One of the reasons I wanted to move to a menu based view switcher was I want to see playlists moved away from the browse folder view and into their own thing where we can better support them. I did some prelim work on EXTM3U support once upon a time, the basic list can be done easily with mpc -f iirc but we'd need to rework the php that creates the client side data for the lists.

Podcasts would be cool too, I wonder how much work it'd be to subscribe to a feed and then have mode pull in the audio automatically and keep track of position, etc.

duracell80, 2020-04-27:
Yeah that would be cool, a separate folder like RADIO.

For now I have my podcasts in RADIO/_Podcasts
[https://github.com/duracell80/MoodeRadio-Get/blob/master/scripts/make-radio-manager/radio-folder.png?raw=true](url)

closed 2020-05-11 by moodeaudio