moode-player/moode #104: upmpdcli out of date

TheOldPresbyope, 2020-05-03:
# Issue

moOde 6.5.1 installs upmpdcli-1.2.16. The current release is 1.4.7. It includes numerous bug fixes, reorganization of code, and new features.

The choices to upgrade are either

## a) Install from packages
I already posted in the Test Team section of the forum a [recipe](http://moodeaudio.org/forum/showthread.php?tid=2451&pid=19451#pid19451) for installing current upmpdcli from JFD's repo. 

or

## b) Build from sources
I will soon post a PR for a modified build_recipe_v2.21.txt which builds libnpupnp-2.2.1, libupnpp-0.18.0, and upmpdcli-1.4.7. I have tested the steps on a fresh install of moOde 6.5.1 and used the results to exercise the local mediaserver and also the Tidal plugin against assorted UPnP Control Points.

## Loose ends

### Spotify
upmpdcli now sports a Spotify plugin which I presume behaves like the Tidal plugin. I have not tested it yet. It depends on the outdated/deprecated libspotify (rather than librespot), a link for which can be found in JFD's documentation.

### OpenHome (or OpenHome Libraries?) and SongCast
There is the possibility of support for both in current upmpdcli but I'm a bit baffled by it all so I've left it alone.

Lttlwing16, 2020-05-06:
I'm seeing the current version of Upmpdcli as 1.4.9-1, which includes the gmusic oauth login features. 

https://framagit.org/medoc92/upmpdcli/-/blob/master/debian/changelog

TheOldPresbyope, 2020-05-06:
It's up to Tim of course, but JFD's website says the released version is 1.4.7.

moodeaudio, 2020-05-06:
I generally only go with production releases from the maintainers. The bump to upmpdci is on the TODO list for moOde 6.6.0 release so maybe by that time 1.4.9 will be production.

moodeaudio, 2020-06-25:
@TheOldPresbyope,

I uploaded 1.4.12 upmpdcli source tarballs and an updated compile recipe to the repo.
https://github.com/moode-player/moode/blob/develop/other/build/build_recipe_v2.23.txt
https://github.com/moode-player/moode/tree/develop/other/upmpdcli

When you get the time, run the compile recipe and see if all is well. I added --disable-spotify to the upmpdcli compile cos I'm not planning on supporting that.

TheOldPresbyope, 2020-06-25:
Compiled fine. I can't test the result until later today.

moodeaudio, 2020-06-29:
Closing due to results from testing that indicate 1.4.12 is an OK upgrade.

closed 2020-06-29 by moodeaudio