moode-player/moode #106: vol.sh glitch issue with USB volume knob

mtnbrit, 2020-05-05:
I and others have found with the USB volume knob function with triggerhappy daemon, sometimes the volume jumps down to a low number if you wang the knob kind of hard and make a large change quickly. I added some debugging to vol.sh and found a lot of error messages about the database being locked. Seems this sqlite3 error is not handled at all so the result gets read into the array anyway and corrupts the variables.

I fixed it by adding a timeout to the sqlite3 select call. I found 1000ms was enough to fix it. Lower values, I could still make it fail. I read that setting any non-zero timeout makes it behave differently by waiting and retrying if the db is busy or locked.

Patch line 56 in /var/www/vol.sh like this
```
RESULT=$(sqlite3 -cmd ".timeout 1000" $SQLDB "SELECT value FROM cfg_system WHERE id IN  ('32', '33', 137)")
```

moodeaudio, 2020-05-06:
The patch works well :-)

Is there a way to catch the return code from sqlite3 in vol.sh and maybe take an exit to avoid the bogus value from being set?

mtnbrit, 2020-05-06:
maybe just exit if result is null, to avoid this

```
++ sqlite3 /var/local/www/db/moode-sqlite3.db 'SELECT value FROM cfg_system WHERE id IN  ('\''32'\'', '\''33'\'', 137)'
Error: database is locked
+ RESULT=
+ readarray -t arr
+ VOLKNOB=
+ VOLMUTE=
+ MPDMAX=
+ REGEX='^[0-9]+$'
+ [[ -up = \-\m\u\t\e ]]
+ [[ -up = \m\u\t\e ]]
+ [[ -up = \-\r\e\s\t\o\r\e ]]
+ [[ -up = \r\e\s\t\o\r\e ]]
+ [[ -up = \-\u\p ]]
tail: /var/log/volsh.log: file truncated
+ LEVEL=1
+ [[ 1 =~ ^[0-9]+$ ]]
+ ((  1 >   ))
/var/www/vol.sh: line 105: ((: 1 >  : syntax error: operand expected (error token is ">  ")
+ ((  1 < 0  ))
+ ((  1 > 100  ))
+ sqlite3 /var/local/www/db/moode-sqlite3.db 'UPDATE cfg_system SET value=1 WHERE id='\''32'\'''
Error: database is locked
+ [[ '' = \1 ]]
+ mpc volume 1

```

moodeaudio, 2020-05-06:
This seems to work as well as the timeout.

```
# Get config settings
RESULT=$(sqlite3 $SQLDB "SELECT value FROM cfg_system WHERE id IN  ('32', '33', 137)")

# Check for empty result due to "Database locked" error
if [[ $RESULT = "" ]]; then
	echo "Empty result from SQL query, exiting"
	exit 1
fi
```

TheOldPresbyope, 2020-05-06:
This last seems to work for me too. The abrupt jumps to 0 are gone no matter how fast I spin the Drok. There remains an occasional odd burble in the rate of change of volume at intermediate rates of rotation of the knob but that may be down to the knob's firmware. Not an issue for me. I turn the knob to get the volume I want to hear, not to watch the numbers roll!

closed 2020-05-11 by moodeaudio