moode-player/moode #110: Feature Request: Multiroom audio

aphilpott, 2020-05-07:
It would be great if Moode supported playing the same music on multiple players in different rooms.

I think using https://github.com/badaix/snapcast would be the best way to achieve this.

I've got a basic prototype of this working on my two Moode players. At the moment my prototype only works for Spotify, and it required me to SSH onto my players to install an start the required software. It would be great if it was part of the Moode image and could be enabled through the config menu in the web UI.

If there's interest in this feature I'd be up for implementing it.

moodeaudio, 2020-05-07:
I've looked at SnapCast b4 and never had the time to get into it. There is another approach that I think might be more reliable and simpler to implement. It's a GStreamer based system that Charlie Laub developed. it's based on ALSA loopback and thus eliminates the need for MPD FIFO pipe or for pipes from other audio applications like shairport-sync, Librespot, Bluetooth, etc.

http://audio.claub.net/GSASysCon.html

https://www.diyaudio.com/forums/pc-based/293266-bash-script-based-streaming-audio-system-client-controller-gstreamer.html

Check it out and let me know what you think.

aphilpott, 2020-05-09:
I've had a brief look into this. My initial impression is that Snapcast requires less setup than Gstreamer/GSASyscon would (see next reply for more detail). But before we decide which is the best option, I think we need to pin down what we're actually trying to achieve here.

I think the key requirements are:

* Ability to play the same audio in sync out of multiple different Moode players
* We need a user-friendly way to dynamically control which players are included in the multiroom set (e.g. ability to go from having the same music in kitchen and dining room during dinner and then switching it to play in the living room after dinner)
* While some of the players on a network are playing in sync, it should be possible to carry on using the rest of the players independently (apart from some obvious restrictions, for example Spotify won't let you stream different things to two instances of librespot running on the same LAN)
* The multiroom system should work for all of the players that Moode supports (e.g. MPD, librespot, shairport-sync, etc.)

There's an interesting question about how we control which rooms audio is playing in. We'd probably want something in the Moode UI, but there are some interesting interactions with some of the players, for example:

* AirPlay already supports multiroom playback i.e. on an Apple device, you can select multiple AirPlay players to stream the same audio to. It would be important not to implement something which interferes with that (e.g. I could imagine adding an option to the Moode UI which lets you stream all the audio from player A to players B and C. But to an Airplay user, the result of that would be that if they select player A on their Apple device, audio actually plays on players A, B and C).

* Spotify doesn't have native multiroom playback, but does let you choose which device to play music from within its app. It would be nice if Spotify users could take advantage of multiroom playback without needing to leave the Spotify app. That's what I managed to get working in my prototype. So for example I have set up my system so that I can synchronise my kitchen and living room, and this is what I see in the Spotify app:
[image]

One other thing which I don't think is a must-have for an initial version of multiroom audio is the ability to have multiple different groups of players in use simultaneously (e.g. one thing playing in both kitchen and dining room, and something else playing in both living room and bedroom).

@moodeaudio any comments on the above thinking?

aphilpott, 2020-05-09:
I've written up some details on what I know about Snapcast and GSASysCon, and what's involved in setting them up:

## Snapcast

In my prototype, I got Snapcast up and running in about 30 mins. Snapcast consists of two packages, snapserver and snapclient, and I just had to:

* install the snapclient package on all my Moodes
* install the snapserver package on one Moode
* change one line of the /etc/snapserver.conf file to tell it to use librespot as the audio source

There's no config to write for the clients. The snapserver automatically detects all of the snapclients on the same LAN (which also makes adding additional Moodes to your network really convenient). There's an android app which you can use to tell the server to just stream to a subset of the clients.

In my prototype, I had my snapserver set up to use librespot as its audio source. But according to the docs, you can quite easily configure ALSA to redirect all audio to snapserver, which would allow us to make this work for all players, not just Spotify, without setting up a separate pipe for each player.

As well as the android app, there's also a JSON RPC API which might make it possible to control which players audio is streamed to from the Moode UI.

## GSASysCon

I haven't actually tried getting GSASysCon up and running yet, but it seems to require more effort to set up, for example:

* writing a config file detailing the IP and port of each of the clients you want the server to stream to
* setting up passwordless SSH between the server and clients

I think the above two points mean the Moode code required to get GSASysCon working would be more complex than for Snapcast. We'd have to generate the required config file, and set up passwordless SSH. Making it so that new Moodes can be seamlessly added to the network would probably be a but fiddly, whereas this Just Works with Snapcast. I'm not sure how changing the set of players that are playing in sync would work - I think we'd have to generate a config file for the set of clients we wanted to include, and then rerun the GSASysCon shell script.

## Conclusion

Overall Snapcast seems simpler, so my recommendation would be to go with that. @moodeaudio what do you think?

moodeaudio, 2020-05-09:
Interesting. If the docs indicate that SnapCast can be fed from alsaloop then it reduces the complexity.

In the Snap world does the Pi running MPD also play in sync with the remote clients?

aphilpott, 2020-05-10:
Looked into this a bit further.

It's not quite true that SnapCast can be fed from alsaloop - it's actually that it's possible to configure ALSA to route all of the sound output of the system into SnapCast's pipe (see the ALSA section of https://github.com/badaix/snapcast/blob/master/doc/player_setup.md). We probably don't want to do that, because of the issue I raised above about interfering with the multiroom features of some players such as Airplay.

To answer your question about whether the Pi running MPD also plays in sync with the remote clients, you can set it up either so that:

* the Pi running MPD just acts as a Snapserver and doesn't play any audio itself, and the other Pis act as Snapclients and actually play audio
* the Pi running MPD acts as both a Snapserver and a Snapclient, so it also plays the music in sync with the rest of the clients. We probably want to do it like this to avoid the need for having an extra Pi

aphilpott, 2020-05-10:
I think we should only implement multiroom for MPD and Spotify, because I think these are the only players on Moode that don't already support multiroom playback:

* Airplay/shairport-sync has native multiroom support
* if you're using Squeezelite, I think you'd be able to get multiroom playback via the Logitech Media Server web interface
* not so sure about uPNP, but from a quick Google looks like there are uPNP servers that support multiroom

I think a nice user experience would be:
* multiroom mode can be enabled/disabled through the audio configuration menu in the Moode web UI
* once multiroom is enabled, MPD users can control which rooms their audio plays in through the main Moode web UI
* Spotify users see an extra device showing up in the Spotify app (called something like "multiroom" by default, but configurable within the Moode config menu). They have to go into the Moode config menu, or use the 3rd party Snapcast app, to change which rooms are included in the multiroom group, but can switch between individual rooms and multiroom within the Spotify app. At the moment, I think it's going to be too difficult to allow Spotify users to specify which rooms are covered by "multiroom" from within the Spotify app.

@moodeaudio what do you think of this proposal?

moodeaudio, 2020-05-10:
I think these multi-room systems for example LMS/Squeeze or iTunes/Airplay are mutually exclusive to one another and would be mutually exclusive to MPD/Snap.

I can envision MPD/Snap multi-room with a nice UI for controlling things which by the way would not be trivial to develop. It would definitely take a significant effort.

The part that I'm not too sure of though is using Snap to do multi-room for the audio renderers since they take over from MPD when a client connects and they overlay the UI because the UI is only for controlling MPD. The other concern is that since SnapCast doesn't support alsaloop things get complex with regard to configuring the renderers to dump their audio data into a fifo pipe.

aphilpott, 2020-05-23:
Yes, I agree about mutual exclusivity of the other multi-room systems and MPD/Snap.

Agree that it would take significant effort to develop a UI for doing MPD+Snap, so we probably don't want to do that unless we're really sure Snap is the way to go.

The part I'm most interested in adding is Snap for the Spotify renderer, because I mainly listen to music through Spotify. You make a good point that this wouldn't be controllable through the UI, but I think that's OK. The way my prototype works is:

* the existing Spotify renderer is running on each of my Moode's. In addition, the one which is running Snapserver is running a second instance of librespot
* most of the time, I have the Snapserver configured to stream to my Kitchen and Living Room, which means I see all my individual rooms, plus "Living Room and Kitchen" in the Spotify app (as in the picture above). This covers most of my use cases
* if I want to change the rooms that Snapserver streams to, I use the Snapcast android app to change that
* so I never need to use the Moode UI to control which rooms music is playing in. The thing it would be nice to do in the Moode UI is start and stop Snapserver and Snapclient

I think adding Snap/Spotify would be a good way of trying out Snap without doing as much development work as MPD/Snap would require. If it went well, we could then think about MPD/Snap.

mgafner, 2020-05-31:
I made a similar prototype / hack for my home. I have a snapcast server & librespot on my homeassistant raspberry pi. In every room i have a raspberry pi with Moode, on which i installed the snapcast clients. Then i changed this files:

> pi@moode-room1:/var/local/www/commandw $ grep snap *
> spotevent.sh:	/bin/systemctl stop snapclient
> spotevent.sh:	/bin/systemctl start snapclient
> spspost.sh:/bin/systemctl start snapclient
> spspre.sh:/bin/systemctl stop snapclient

so, if i use librespot in a single room, it stops the snapclient and play only this room. In any other case, it casts the sound of my central snapcast server. I am using the snapcast android app to control the volumes.

i would be happy to see snapcast integrated in Moode ;-)

moodeaudio, 2020-06-03:
I don't have the bandwidth but maybe another developer thats interested and motivated will take on the challenge of integrating this into moOde.

closed 2020-06-03 by moodeaudio

tjengbudi, 2023-11-24:
hello,
i want try to implement snapcast into moode. however cause of my lack information about moode i want to ask something.
for my test i used snapcast server in moode. and client in my android and it works. i used loopback for source for snapcast server.

however there is some delay cause in moode that have snapserver, the sound is not from snapclient and used loopback
so... can i cut the tie of alsaloopback to snapcast server. and the sound get from snapchat client? so the device used as snapcast server and snapcast client

can this works in moode?