moode-player/moode #112: Bluetooth audio latency increases over time

TeChn4K, 2020-05-08:
Hi Moode team,

I often use Moode with the Bluetooth renderer, paired with my laptop, audio output to local and buffer set to 20ms.

I notice that almost every time, Bluetooth audio latency increases from unnoticeable to maybe 300 / 500ms. Some times it happens at the very beginning of my session (less than one minute playback), but each time after a longer period (45min / 1h).
I have to disconnect & reconnect Bluetooth to reset the latency.

Issue happens from at least Moode 6.4 to 6.5.2
Do you have any idea about how to track down this issue?

Configuration :
- Moode Player 6.5.2
- ALLO USBridge Sig
- Wifi & Bluetooth dongles
- USB audio output

moodeaudio, 2020-05-08:
What is the format of the audio file?
How are you able to measure the latency?

TeChn4K, 2020-05-09:
I use Moode and bluetooth to send audio from my computer (Youtube or VLC).
I did not measure the latency precisely. I'm able to see it when I watch video and notice the lip sync delay.

Audio information details : 
```
Input / Output
Source: Bluetooth stream
Encoded at: Unknown
Decoded to: 16 bit, 44.1 kHz, Stereo, VBR
Destination: Local
Output rate: 32 bit, 44.1 kHz, Stereo, 2.822 Mbps

DSP operations 
Volume ctl: Software (MPD 32-bit float with dither)
Resampling: n/a
Polarity inv: n/a
Crossfade: n/a
Crossfeed: n/a
Parametric EQ: n/a
Graphic EQ: n/a
Replaygain: n/a
Normalize vol: n/a
Chip options: None

Audio Device
Device: USB audio device
Chip: 
Interface: USB
Formats: S32_LE
Platform: Allo USBridge SIG [CM3+ Lite 1GB v1.0]
```

moodeaudio, 2020-05-10:
The only option in moOde that could affect BT latency is the buffer length setting in BlueZ config. 

The lib-sync issue with YT video has been around for ages but I don't recall any issues when using Airplay. Maybe give that a try.

closed 2020-05-11 by moodeaudio

TeChn4K, 2020-05-11:
I've tried with buffer length set to 20, 40 and 60ms. For me, these latencies are almost unnoticeable when you watch a video of somebody speaking.

Just to be clear: most of the time, bluetooth latency is perfectly OK and in perfect adequacy with the BlueZ config. You can watch a YT video with no issue at all.

What I'm talking about is a shift issue of the latency.
One moment everything is fine, latency is OK, then randomly the latency shift instantly to something unbearable. I'm forced to switch on/off bluetooth connection to reset it.

moodeaudio, 2020-05-11:
I don't know what could be causing the latency to suddenly shift or how to troubleshoot something like this. The buffer length setting is a bluez-alsa option so maybe posting the issue to bluez-alsa repo could shed some light on what's happening.

```
pi@rp3:~ $ cat /etc/bluealsaaplay.conf
# bluealsaaplay.conf
AUDIODEV=plughw:0,0
BUFFERTIME=500000
```