moode-player/moode #113: Removing Bluetooth dongle hangs moodeaudio on startup

taiebot, 2020-05-10:
Hi.

I could not figure why moodeaudio was not starting. I did check moodeutl -l and saw that it was hanging at 
20200510 175358 worker: Bluetooth (available)
20200510 175358 worker: Bluetooth (started)

weird because i had removed my bluetooth usb dongle. By reputting the dongle i managed to get access to the device.

moodeaudio, 2020-05-10:
If you open Audio Config not the warning under the Bluetooth ON/OFF button.

"WARNING: A Bluetooth adapter must be physically attached and enabled before turning this ON otherwise moOde startup will fail."

taiebot, 2020-05-10:
As i could not boot I did not see the comments.

moodeaudio, 2020-05-10:
Right, but the warning would have been plainly visible when you first turned Bluetooth on in Audio Config.

The reason for the warning is that I have not yet found a way to detect the presence or absence of a Bluetooth adapter during moOde startup. If there is a way to do this I'll be happy to test and implement if it works.

closed 2020-05-11 by moodeaudio

virtadpt, 2020-06-16:
How do we fix it, though?  Stopping and disabling bluealsa.service isn't enough.  Editing /etc/mpd.conf to delete the "ALSA Bluetooth" stanza isn't sufficient.

virtadpt, 2020-06-16:
I think I figured it out.

`sudo sqlite3 /var/local/www/db/moode-sqlite3.db`

`sqlite> update cfg_system set value=0 where id=78;`

`sqlite> .quit`

Then reboot.

moodeaudio, 2020-06-16:
Prolly best not to use the id since that may change in future. Try below.

```
sqlite3 /var/local/www/db/moode-sqlite3.db "update cfg_system set value='0' where param='btsvc'"
```