moode-player/moode #119: integrityCheck() enhancement perhaps

TheOldPresbyope, 2020-05-20:
This is the result of a self-inflicted wound. I had modified nginx.conf on a player in preparation for exploring IPv6 issues. In the throes of old-timers disease, I forgot I did it. Days later, I rebooted this particular player and it misbehaved. Checking moode.log, I see

> 20200520 101557 worker: Integrity check (failed:3)

Ok, so looking at playerlib.php, I see that code 3 means some file failed the hash test. I then read the cfg_hash table to see what files are being checked and had to guess that I was tripping over the forgotten mod to nginx.conf.

Had the file been marked in the cfg_hash table as action="warning" rather than action="exit", it would have been named in the workerLog entry created in the integrityCheck function. Couldn't the same be done for action="exit" files? (even though the actual log entry for an action="exit" is generated elsewhere.)

moodeaudio, 2020-05-20:
The Integrity Check system in moOde is designed to reduce trouble reports that are the result of certain files being hacked manually or as a result of installation of 3rd party "tweaking" software.

Generally the NGINX and PHP configs are only updated after a major new release of these packages and thus they are marked 'exit'. The other files marked 'warning' or 'ignore' should also be marked 'exit' since they are critical files but there is still a need to have a certain level of hackability that allows users to experiment.

TheOldPresbyope, 2020-05-20:
Not questioning the usefulness of the function, just asking if it could be more chatty.

To continue testing, I've now turned it into a Cole Porter function ("Anything Goes").

closed 2020-05-20 by TheOldPresbyope