moode-player/moode #123: [question] about covertart.php and the use of $_SESSION['musicroot_ext']

bitkeeper, 2020-05-22:
If cover art is requested there are two possible sources:

1. embedded art in the song
2. artwork in in a file

For first the following request delivers and image:
http://moodedev/coverart.php/NAS%2FArtist%2FAlbums%2FlSong.flac

For the second the following request:
http://moodedev/coverart.php/NAS%2FArtist%2FAlbums%2FlSong.flac
Results in a redirect:
http://moodedev/3063407/NAS/Artist/Album/cover.jpg
 
After searching the code is seems that the only reason for the directory 3063407 (current `$_SESSION['musicroot_ext']`) to exists is to serve cover files. 
It is not directly clear why the construction with symbol link with randomw name is required ?
Is there any reason to treat this different than with the embbed file, by just serving it from disk (example impl below)?  

If not the entire `$_SESSION['musicroot_ext']` construction and the directory linking could be removed, which would cleanup the code. And also makes working with development proxy like gulp browserSyn a lot easier.

The redirect is implemented as:
```php
$path = '/' . $GLOBALS['musicroot_ext'] . substr($path, strlen(MPD_MUSICROOT)-1);
$path = str_replace('#', '%23', $path);
header('Location: ' . $path);
exit(0);
```
Direct served from disk:
```php
$mimeType = 'image/'.$ext;
$filesize = filesize($path);
$fh = fopen($path, 'rb');
$imageData = fread($fh, $filesize);
fclose($fh);
outImage($mimeType, $imageData);
break;
```

bitkeeper, 2020-05-22:
Ok clear.In that case I will make a PR with the stuff removed and the direct serve from disk.

I'm currently creating a [gulp](https://gulpjs.com/) project file to automated some stuff for easier frontend development of moOde like:

- bundling all css to one file
- bundling all js to one file
- minify everything
- create sourcemaps to still make easy debugging possible
- user browserSync with as test server for frontend (with automatic reloading if aresource changes)
- do your frontend dev on your pc while still using the backend on the pi, by using the proxy in browserSync
- building all above steps in a single step
- deploy target from source tree to /var/www

So now and then I hit some quirks in the code that prevent the usage of modern web app tooling, that is why the question about musicroot_ext.

moodeaudio, 2020-05-23:
I merged PR #125 so lets close this issue.

Feel free to open another regarding the gulp stuff etc.

closed 2020-05-23 by moodeaudio