moode-player/moode #132: Spotify playback always misses the first 500-1000ms of a track on first play

ahydronous, 2020-05-31:
Whenever I play a playlist, or start playing a new track, the track will start playing but miss the first 500-1000ms of audio, presumably because it is buffering.
If I start the track again by pressing <, the track plays perfectly.
The problem also happens when playing playlists or albums, which is extremely annoying when expecting gapless playback.

moodeaudio, 2020-05-31:
You should prolly post this in the librespot repo.
https://github.com/librespot-org/librespot

The version of librespot used in the current release of moOde is v0.1.1 which does not support gapless play. The librespot dev's have pushed commits that enable gapless play but they have not published the release yet. When they do that I'll include it in moOde.

closed 2020-05-31 by moodeaudio

ahydronous, 2020-06-09:
Cool, thanks for the info!