moode-player/moode #133: Stream on other port as 80 not working?

tiborepcek, 2020-06-03:
Hi and thank you for MoOde Player.

I would like to ask, why stream on URL https://stream.funradio.sk:18443/fun64.mp3 does not work on MoOde Player on RPi 3 B+, but in web browser it works.

All other streams on port 80 work just fine on MoOde Player on RPi 3 B+.

Any advice?

Thank you.

moodeaudio, 2020-06-03:
I can repro and it looks like some kind of certificate issue. The error from MPD is below. If you can view the cert in your Browser then maybe try posting the issue to MPD or Raspbian maintainers. 

```
Failed to decode https://stream.funradio.sk:18443/fun64.mp3; CURL failed: server certificate verification failed. CAfile: none CRLfile: none
```

moodeaudio, 2020-06-03:
Most likely not a moOde issue
https://github.com/MusicPlayerDaemon/MPD/issues/887

closed 2020-06-03 by moodeaudio