moode-player/moode #136: Radio view options

moodeaudio, 2020-06-03:
No that the first cut of import/export has been committed I'm back to updating the newly added columns in cfg_radio. I'm about half done and have enough updated stations to begin developing a new Radio view.

The new columns (tags) are:
genre
broadcaster
language
country
region
bitrate
format

The bitrate and format  tags will allow displaying this for each station as a badge, text or "HD" similar to what's already implemented for Tag and Album views.

Couple questions:

1. What bitrate would be considered HD for radio stations? 
2. What are some options for presenting Radio view?

monkeypox, 2020-06-04:
I don’t really use the radio function but...

#1 I would consider 256kbps as a high bitrate for radio but it might be confusing if you use the same “hd” badge for different meanings depending on the view.

#2 do you think people would select multiple options from the different categories? It seems like it’d be cool but the number of built-in stations really isn’t that large so it might be overkill. If multiple options are desirable then I think it’d be cool to let the user create named, saved filters (e.g. genre: jazz, bitrate:256kbps) that would show up at the top of the list of stations as folders and then when clicked would show the subset of stations that matched that criteria.

If there are going to still be < 100 stations then you could also just add user facing folder creation. You could use a global var to detect state so when a user selects an item it diverts from normal function and uses .toggleClass() on the li element and add some css to show it as selected), then have save / cancel buttons overlaid on the playbar. Cancel just removes the class from all elements, save pops up a dialog. This is basically how I was implementing playlist cut/paste and iirc it was pretty easy though I haven’t looked at that code in a while. It’d probably be easiest to symlink the station files into a physical directory because then the existing code should just work. I don’t think you’d want to actually move them because then you’d have to deal with people wanting to move them back, or from one folder to another, or not being able to have stations in more than one folder, etc.

Both would be useful but the former would be better for a large number of stations whereas the latter might be more generally useful for users who want to organize stations.

I don’t have any radio folders so this might already be a thing but moode should save the state of the radio view such that on refresh it loads the folder the user was last in instead of the default “home” one.

moodeaudio, 2020-06-05:
Good point. The HD designation is more or less agreed on for Albums/songs and HD Radio is a specific service and something altogether different as I've learned.

 I've seen the term Hi-Res used for streaming so maybe thats an option. Probably > 128K could be considered Hi-Res but there are two tiers, lossy and lossless for example 320K AAC and 24/96 FLAC. it would be nice to distinguish them.

There are around 200 stations in the stock collection and I'm trying to avoid folders and instead have the view based on the newly included metadata.

manderso7, 2020-07-28:
New user, thoroughly enjoying the software.
Is there a possibility of an option for the to resize the icons? 
Also, could there be a list of radio stations?
Perhaps put a link to the place to find more stations. I know I saw a link for a list of stations in the forums, but can't find it currently.

Probably more questions/suggestions on radio stations in the future.

moodeaudio, 2020-07-28:
You need to provide more context around your questions for example:

- Are you asking for larger or smaller button icons, on which screens, what's the usage scenario that would benefit, etc.?
- What do you mean by a list of radio stations? The stations are already listed in Radio view.

manderso7, 2020-07-28:
I'll try to be clearer. 
I'd like the ability to resize the covers for the radio stations, preferably w/ a slider, in order to fit more on the screen. Also, a list of radio stations meaning put the stations in list format
[image]
, instead of just cover format:
[image]

moodeaudio, 2020-07-28:
The Appearance option "Thumbnail columns" is used to increase or decrease the number of thumbnails shown.

What advantage is there to having a list format in addition to the existing grid format?

The new Radio view coming in moOde 7 series is going to be based entirely on the newly added station metadata and not dependent on an underlying folder structure. In fact sub-folders for radio stations are not going to supported.

manderso7, 2020-07-28:
It's not just me, sometimes a list is easier to navigate.
http://moodeaudio.org/forum/showthread.php?tid=2832

gbh-uk, 2020-07-28:
To answer your question 2: Please consider making all of these new metadata 'columns' you mention capable of being hidden.
I don't like speaking for other people, and wouldn't want to limit a perceived enhancement based on my views, but the current alphabetical list is more than adequate imho.
I'd prefer to be able to hide the clutter of metadata that's neither interesting nor useful (to me).

moodeaudio, 2020-07-28:
We've had a many requests over the years for a Radio view that can be presented by Genre, Bitrate, Audio format, Language, Region HiDef badge, and so on.

The first step was to code all the built in stations with metadata and then reflect that in the station add/edit dialogs.

It's still up in the air as to how the new Radio view will operate because nothing has been mocked up yet. If I were to guess it will prolly end up being a grid but with an easy to use button bar for filtering. Something like that.

moodeaudio, 2020-09-07:
The new Radio view is done and will be included in upcoming moOde 7 :-)

closed 2020-09-07 by moodeaudio