moode-player/moode #137: how do you provide cover art over a http stream to mdp?

lachlan-00, 2020-06-05:
Hey, i know this isn't really specific to moode but i really like using moode with my ampache server and was wondering if you could help get the art displaying?

I play these stream urls to the mpd server
https://my.ampache.server/play/ssid/%ampachesessionID%/type/song/oid/105444/uid/2/name/KoRN%20-%20Counting.mp3

that plays totally fine, just without the art.
[image]

The art exists and even if it didn't i'd like to provide a backup/fallback image in the server.
[image]

I'd really like to make the changes to Ampache to benefit everyone but i'm just not really sure how players like moode use/check for images.

lachlan-00, 2020-06-05:
secondarily can i provide a server name to the client?
[image]

lachlan-00, 2020-06-05:
sorry for the spam but it would also be cool for the playlist to load the details before the track is played
[image]

moodeaudio, 2020-06-05:
Our project only provides support for moOde's built in WebUI.

closed 2020-06-05 by moodeaudio

lachlan-00, 2020-06-07:
This question is about how to get URLs that display things in the moode webUI but okay