moode-player/moode #138: function genWpaPSK fails on special chars in SSID

TheOldPresbyope, 2020-06-15:
As reported in the forum by user ab.basu (post [http://moodeaudio.org/forum/showthread.php?tid=1827&pid=21225#pid21225](url), moOde fails to associate with an AP if its SSID contains the dollar sign ($), which is a legal character in this context (see below).

The problem lies in /var/www/inc/playerlib.php in function genWpaPSK in this line (#2706 in my 6.5.2)

`	$result = sysCmd('wpa_passphrase "' . $ssid . '" < /tmp/passphrase');`

Bash will interpret special characters in strings contained within double quotation marks.

I tried hacking this line to change the double quotation marks to escaped single quotes (\\') which seemed to work, but I'm not the PHP guru.

Special characters: 

1) The SSID can be 0-32 octets of unspecified encoding according to the standard, so naming is a tricky business. Various router/AP mfgrs set their own limitations

2) maybe there's a better way to escape these in PHP

3) there's also the matter of writing and reading special characters from SQLITE3 dbs. I'm no expert at this either.

TheOldPresbyope, 2020-06-15:
PS - I probably should have added another note:

4. user input: I haven't checked how you capture and validate the SSID entry in the Network Config panel.

moodeaudio, 2020-06-16:
Theres a bug in the validation pattern so its not really functioning correctly. Heres the fixed pattern but its only applicable to manually entered SSID. For SSID's in the scan list there is no validation.

Only allow alphanumeric, dash and underscore. No embedded spaces, 3 to 32 chars.
```
pattern="[A-Za-z0-9-_]{3,32}"
```
Technically there are other invalid formats for SSID. It can't start with certain chars, etc.

moodeaudio, 2020-06-16:
I think originally double quotes were used to enclose the SSID because some users reported using single quotes in their SSID's. Something like that.

TheOldPresbyope, 2020-06-16:
Unfortunately, the IEEE spec is hidden behind a paywall. Here's the Wikipedia interpretation

> SSIDs are usually customizable.[8] These SSIDs can be zero to 32 octets (32 bytes) long,[9] and are, for convenience, usually in a natural language, such as English. The 802.11 standards prior to the 2012 edition did not define any particular encoding/representation for SSIDs, which were expected to be treated and handled as an arbitrary sequence of 0–32 octets that are not limited to printable characters. The IEEE 802.11-2012 defines a tag that the SSID is UTF-8 encoded and when interpreting could contain any non-ISO basic Latin characters within it.[10] _**Wireless network stacks must still be prepared to handle arbitrary values in the SSID field**_. 

I added the emphasis.

Vendors and software app writers may impose limitations but the spec doesn't. Cisco, for example, has a bunch

>  The SSID can consist of up to 32 alphanumeric, case-sensitive, characters.
>  The first character cannot be the !, #, or ; character.
>  The +, ], /, ", TAB, and trailing spaces are invalid characters for SSIDs. 

And many AP/client software which uses bash or busybox may have issues---which is why we're here:)

I think we just have to do what's reasonable and then be explicit about any limitations in order to manage users' expectations.

Maybe I'm overly sensitive to this issue, but it's with cause. Some years ago, I bought a WiFi-enabled scale which I couldn't associate with my router until I created a guest network with a totally dumbed-down SSID.

moodeaudio, 2020-06-17:
Right, unfortunately there is no standard for SSID other than its length and given that the current approach in moOde seems to be working fine for the vast majority of users I would propose not making any changes.

If we had a lot of reports of special character SSID's causing issues then for sure we would make changes.

moodeaudio, 2020-08-06:
If we get more trouble reports relating to this issue I'll reopen.

closed 2020-08-06 by moodeaudio

tiagovaz, 2023-01-06:
> If we get more trouble reports relating to this issue I'll reopen.

So please do, I'm having trouble setting a SSID manually via the web GUI when the SSID has a '.' on its name. Thanks!

moodeaudio, 2023-01-06:
Do you mean the SSID contains a period? 

Maybe provide an example SSID that has the character(s) that currently are not allowed by the input field.

tiagovaz, 2023-01-06:
Yes, a period. It's tt2.4

moodeaudio, 2023-01-06:
Edit the file /var/www/templates/net-config.html and add the period to the pattern.

```
		<div class="modal-body">
			<div class="control-group">
				<label class="control-label" for="wlan0otherssid">SSID</label>
				<div class="controls">
					<input class="config-input-large config-modal-input" type="text" pattern="[A-Za-z0-9\- .]{3,32}" id="wlan0otherssid" name="wlan0otherssid" value="" autofocus>
					<a aria-label="Help" class="config-info-toggle" data-cmd="info-manualssid" href="#notarget"><i class="fas fa-info-circle"></i></a>
					<span id="info-manualssid" class="config-help-info">
                    	Length 2-32 chars.
                    </span>
				</div>
			</div>
		</div>

```

Confirm whether or not it works for you and if so I'll update the file for the next release.

tiagovaz, 2023-01-07:
Yes, it does work. Thanks for the quick fix! You may want to fix the info/error message as well: "Length 2-32 chars". It seems it should be 3-32 instead.

Moode is wonderful.

Bests,