moode-player/moode #140: Disable nginx access log

rroblik, 2020-06-24:
Hi,

Thanks for the hard work done on this project ; it work pretty well on my Rpi 3+.

As this device is using SD cards OOB ; it's strongly recommanded to limit the number of I/O operations.

As see on my Pi that the moode nginx configuration write a lot in `access.log` file. To my mind that is useless (only `error.log` may be usefull). 

Maybe this behavior must be configured from UI ? Anyway logging must be disabled be default [here](https://github.com/moode-player/moode/blob/a1829239247062f5b80a881742acbfee8a826edc/etc/nginx/nginx.conf#L42).

Other solution can be to embed installation of [log2ram](https://github.com/azlux/log2ram) for Pi image. 

Best regards

moodeaudio, 2020-06-24:
There have been very few reports in our Forum over the years of SDCard failures so I'm not inclined to make any changes but I agree that the access log is not really useful.

How to disable it?

moodeaudio, 2020-06-24:
nvm, the setting is simply "off"

closed 2020-06-24 by moodeaudio

monkeypox, 2020-06-25:
Trying log2ram out. 🤔

I don’t know if there’s any real benefit but I have heard people prefer archlinux based players so this might be an interesting option for moode to offer just from an acquiring users perspective.

moodeaudio, 2020-06-25:
Whats the benefit to audio of archlinux over raspberry pi OS?