moode-player/moode #141: Load available mpd versions from config file

bitkeeper, 2020-06-28:
When the use MPD get gets a version bump, the source of needs to be changed snd-config.php.

Is it an idea to load the available versions of MPD from a config file ?
This will decouple the available mpd version from the snd-config. Also makes it easier to tinker arround with multiple mpd versions.

For example /var/local/www/mpdversions.cfg, with the content:
```
# This config contains the selectable mpd versions for moOde
0.21.24;0.21.24 (Default)
0.22~git;0.22~git (Test)
```
The first is the version for binary name and the second a label. Separated by a ';'. 
Lines with # or empty are ignored.

moodeaudio, 2020-06-28:
Thats a good idea. 

The config file should prolly be named mpd_versions.conf rather than mpdversion.cfg

bitkeeper, 2020-06-28:
Renamed config file to mp_versions.conf in PR #142.

moodeaudio, 2020-06-28:
PR accepted.

closed 2020-06-28 by moodeaudio