moode-player/moode #145: minor changes to audio in Raspberry Pi OS break detection of MPD audio device type

TheOldPresbyope, 2020-06-29:
re [my string of posts to the forum](http://moodeaudio.org/forum/showthread.php?tid=2687&pid=21728#pid21728):

The change from Raspbian to Raspberry Pi OS includes minor changes to audio which break function getDeviceNames () in playerlib.php. In particular, I can't detect and set an MPD audio device type, either onboard or USB.

I hacked the function to test both for 'ALSA' (Raspbian) and 'Headphones' (Raspberry Pi OS). With this mod, I could update/upgrade moOde 6.5.2, which gets me to
```
Raspbian 10.4 | 4.19.118-v7+ #1311 | armv7l (32-bit)
```
and still detect and set my USB dac.

I can also rpi-update to get to the current bleeding edge kernel 5.4.49-v7+ and still detect and set my USB dac.

All alternative renderers seem to work in limited testing.

I do not include proposed code changes here because I'm uncertain about the dichotomy of HDMI socket and the headphone jack described in the blog entry reference in my string of posts on the forum.

moodeaudio, 2020-06-29:
I did a search of the code base and it looks like the only change thats needed is to add tests for $card0 == "Headphones" to function getDeviceNames () in playerlib.php

TheOldPresbyope, 2020-06-29:
That was all I needed but some folks use the HDMI port and I'm not sure how that figures into this new scheme.

moodeaudio, 2020-07-03:
Testing on my end indicates the change I mentioned earlier is all thats needed. If HDMI issues arise in the field after 6.6.0 release then we can revisit.

closed 2020-07-03 by moodeaudio