moode-player/moode #147: DLNA Media Browser mischief

TheOldPresbyope, 2020-07-09:
[Warning - this is a long-winded discussion with no proposed solution at the end of it.]

## Background

The DLNA Media Browser (djmount) is an user-enabled feature in moOde intended to make tracks on UPNP/DLNA Media Servers accessible via the moOdeUI.

To the casual user it works as intended. Enable the browser, create a moOde music source from a UPnP/DLNA server, update the Library, and the source with its albums and tracks appears in the various Library views.

## Problem summary

Updating the moOde Library can bog down, cover files aren't used to create thumbnails and cover displays, and track entries may be replicated _ad naseum_.

## Under the hood

From the djmount sourceforge.net page: 

> Djmount discovers automatically all UPnP AV Media Servers on the network, and make the content available in a directory tree. All shared files (e.g. Audio or Video files) are directly visible and can be played using your favorite media player.

This means literally _all_. The content of every UPnP AV Media Server on the network is made available under /mnt/UPNP in the moOde player, whether or not moOde music source entries have been created for them. This, in turn, means that the script /var/www/command/listall.sh walks all the directories in search of music tracks whether or not its findings will be relevant to the moOde library. (As an aside, the mounts persist even after the DLNA Media Browser is disabled.)

The directory tree mounted by djmount is imaginary. That is to say, it consists of various "searchable" track paths synthesized by djmount based its UPnP queries to the servers. They don't exist on the servers.

To demonstrate, I spun up a miniDLNA server on a DietPi host with a stock configuration file and just two music albums. `/mnt/dietpi_userdata` is the root directory of the server. Here's the directory tree structure:

```dietpi@DietPi:/$ tree mnt
mnt
├── dietpi_userdata
│   ├── Music
│   │   ├── SOUL
│   │   │   ├── Aretha Franklin-Aretha Lady Soul
│   │   │   │   ├── 01.Chain of Fools.flac
│   │   │   │   ├── 02.Money Wont Change You.flac
│   │   │   │   ├── 03.People Get Ready.flac
│   │   │   │   ├── 04.Niki Hoeky.flac
│   │   │   │   ├── 05.(You Make Me Feel Like) A Natural Woman.flac
│   │   │   │   ├── 06.Since Youve Been Gone (Sweet Sweet Baby).flac
│   │   │   │   ├── 07.Good to Me As I Am to You.flac
│   │   │   │   ├── 08.Come Back Baby.flac
│   │   │   │   ├── 09.Groovin.flac
│   │   │   │   ├── 10.Aint No Way.flac
│   │   │   │   └── Folder.jpg
│   │   │   └── Aretha Franklin-I Never Loved A Man The Way I Love You
│   │   │       ├── 01.Respect.flac
│   │   │       ├── 02.Drowning In My Own Tears.flac
│   │   │       ├── 03.I Never Loved A Man (The Way I Love You).flac
│   │   │       ├── 04.Soul Serenade.flac
│   │   │       ├── 05.Dont Let Me Lose This Dream.flac
│   │   │       ├── 06.Baby, Baby, Baby.flac
│   │   │       ├── 07.Dr. Feelgood (Love Is A Serious Business).flac
│   │   │       ├── 08.Good Times.flac
│   │   │       ├── 09.Do Right Woman - Do Right Man.flac
│   │   │       ├── 10.Save Me.flac
│   │   │       ├── 11.A Change Is Gonna Come.flac
│   │   │       └── Folder.jpg
│   │   └── fourdee_tech.ogg
│   ├── Pictures
│   ├── Video
│   └── downloads
```
Note: I truncated the tree to remove placeholder entries for ftp_client, etc., which are empty.

Here's the corresponding directory tree structure on a moOde player with the DLNA Media Browser enabled:
```
pi@moode3A:/mnt $ tree UPNP
UPNP
├── devices
├── DietPi MiniDLNA
│   ├── Browse Folders
│   │   ├── Music
│   │   │   ├── fourdee_tech.ogg
[... 324 lines cut ...]
│       ├── Folders
│       │   └── _search
│       │       ├── search_capabilities
│       │       └── search_help.txt
│       ├── Recently Added
│       └── _search
│           ├── search_capabilities
│           └── search_help.txt
├── FIOSTV_F45FD41DF9D4
└── Moode4B1 DLNA
```
Notes: 
1. "Moode4B1 DLNA" is the DLNA Server enabled on another moOde player. I truncated the tree here. 
2. "FIOSTV..." is our cable box. Had I plugged in a USB drive with music tracks it would have been enumerated too.
3. The Folder.jpg files present in the source directory are not present in the /mnt/UPNP tree.
4. Those _search directories tell the client what searches can be conducted on the fly.

## The consequences of this mischief

When a moOde UPnP music source is created for the DietPi media server and the library is updated:

1. MPD adds each track to its database as many times as it appears in the synthetic directory tree structure---9 times in this demonstration. This is not apparent in the Folder view of the Library but becomes immediately apparent in the Tag view when an artist or an album is selected. (The problem gets much worse with some UPnP media servers. I have an instance of minimserver running on another host. With its default configuration, it creates nearly three dozen different paths to the same file. To keep things simple I chose not to include it in this demonstration.)
2. The album cover files are not exposed by djmount so moOde can't create thumbnails from them and can't display full covers if that's the only choice either.

Things can be even more complicated if several moOde players have both their UPnP Media Servers and their DLNA Media Browsers enabled, but I'll leave that exploration to the interested student.

moodeaudio, 2020-07-10:
UPnP A/V is really messy and is one of the reasons I dropped djmount some years back only to add it back in last year. It's no surprise that Roon invented their own protocol rather than use UPnP A/V.

In any case, I think when creating the music source in Library you have to add the sub-folders to the path for example ".../Music/Album" to keep MPD from indexing the whole UPnP folder tree which contains all those duplicate paths. And yes, IIRC only embedded art will work since as you pointed out the djmount tree doesn't list cover art files.

When we were using the MPD file list as input to the thumbnail generator it was already a filtered list based on the source path ".../Music/Album". Now we are using listall.sh which allows us to generate thumbnail images in parallel with MPD db update running but its scanning /mnt/UPNP which is the entire tree ????

Maybe we need to modify listall.sh to incorporate the source path "...Music/Album" or whatever user enters. Then it would be scanning /mnt/UPNP/DietPi MiniDLNA/Music/Album.

TheOldPresbyope, 2020-07-10:
For sure it's messy. I'm amazed at how open-ended the specs are, I suppose because it was envisioned humans would be doing the browsing. (Besides, my boss used to say that standards committees reach consensus by being vague.)

Certainly we need to prune the tree for both MPD and listall.sh.

The trouble with pre-defining a path such as "../Music/Album" is that it varies with the UPnP/DLNA Media Server implementation. I see quite different results for the instances I've spun up of miniDLNA, minimserver, and Universal Media Server (if I get a chance, I'll try Kodi later today just for grins).

Because of this, I think many users would be challenged to enter the right path as well.

What a muddle. If I get any inspiration I'll add another comment.

moodeaudio, 2020-07-10:
IIRC there was a lengthy discussion thread in the Forum about the UPnP browser and what the expectations were as far as the path being specified so that it only contains the list of Albums. It definitely required some in depth knowledge on the part of the user to know what the particular media server was using as names for the folders.

We can't hard code anything but I suppose we could add some help text explaining what is required when specifying a path for a UPnP media server.

TheOldPresbyope, 2020-07-11:
Argh. Turns out I've naively "discovered" the same issues already discussed three years ago by Francois concerning his work which became the [MPD database UPnP plugin](https://www.lesbonscomptes.com/pages/mpd-upnp.html).

I suspect any explanatory help text will have to be backed up with more detailed explanatory material on the forum.

moodeaudio, 2020-08-06:
I looked at the help for UPnP music source in Library config and I'm inclined to leave it as-is. It directs the user to SCAN into the folder tree and thus leaves it up to the user to determine which particular folder to use as the music source. I remember that there is no standard name for the "Albums" folder in UPnP media servers and so we can't code for it.
```
"For UPnP folders, select a folder then SCAN to display its sub-folders." 
```

closed 2020-08-06 by moodeaudio