moode-player/moode #150: Volume can be scrolled using mouse wheel below zero to -1

TheMetalHead, 2020-07-19:
Moode v6.7.0

There is an issue when using the mouse wheel to change the volume.

To re-create the issue select a stream to play, set the volume to below 5 and then using the mouse wheel, quickly scroll down to zero. The volume will go to -1.
Clicking on the + and - indicators are ok.

moodeaudio, 2020-07-19:
Thats odd. I don't have a mouse to be able to repro but try to verify what the MPD volume is when the knob shows -1.

mpc volume

TheMetalHead, 2020-07-19:
Knob at 4% mpc volume shows 4%
Scroll the knob to -1 mpc volume shows 0%

bitkeeper, 2020-07-20:
Can reproduce with both Firefox and Chrome.

bitkeeper, 2020-07-20:
@TheMetalHead the issues should be fixed in upcoming release 6.7.1.

moodeaudio, 2020-07-20:
PR to address this has been merged.

closed 2020-07-20 by moodeaudio

TheMetalHead, 2020-07-20:
Also changed playerlib.min.js
Confirmed the fix is good on v6.7.0 ok.
Thank you @bitkeeper and @moodeaudio.

moodeaudio, 2020-07-20:
Oops, I just pushed the min

Thx for confirming the fix :-)